OrcaSlicer/OrcaSlicer #5860: There is an issue running Orca slicer AppImage on Ubuntu 24.04

noamix, 2024-06-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer_Linux_V2.1.0.AppImage

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04 fresh install

### Additional system information

# Detalles del informe del sistema
---

## Detalles del informe
- **Fecha generada:**                              2024-06-25 12:41:47

## Información de hardware:
- **Modelo de hardware:**                          ASRock A320M-HDV R4.0
- **Memoria:**                                     16,0 GiB
- **Procesador:**                                  AMD A6-9500 RADEON R5, 8 COMPUTE CORES 2C+6G × 2
- **Gráficos:**                                   NVIDIA GeForce GT 710
- **Capacidad del disco:**                         480,1 GB

## Información de software:
- **Versión de firmware:**                         P3.70
- **Nombre del SO:**                               Ubuntu 24.04 LTS
- **Construcción de SO:**                          (null)
- **Tipo de SO:**                                  64 bits
- **Versión de GNOME:**                            46
- **Sistema de ventanas:**                         X11
- **Versión del núcleo:**                          Linux 6.8.0-35-generic

### Printer

ender 3v2

### How to reproduce

leo@leo-home:~/Descargas$ chmod +x OrcaSlicer.AppImage
leo@leo-home:~/Descargas$ ./ OrcaSlicer.AppImage
bash: ./: Es un directorio
leo@leo-home:~/Descargas$ ./OrcaSlicer.AppImage
/tmp/.mount_OrcaSlu1xsQz/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Actual results

I can't get Orca slicer to run on Ubuntu 24.04

### Expected results

orca slicer must run

### Project file & Debug log uploads

./OrcaSlicer.AppImage
/tmp/.mount_OrcaSlu1xsQz/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Checklist of files to include

- [ ] Log file
- [ ] Project file

john-parton, 2024-06-25:
Note that you cannot even run `sudo apt install libwebkit2gtk-4.0-dev` to resolve, because that package no longer exists in Ubuntu 24.04. They've upgraded it to `libwebkit2gtk-4.1-dev`.

I don't know much about AppImage, but my general gut feeling is that a flatpak wouldn't have this same issue.

closed 2024-07-01 by SoftFever

luigicampanaro, 2024-09-24:
This worked for me:

`./OrcaSlicer_Linux_Ubuntu2404_V2.2.0-beta2.AppImage --appimage-extract
`
`cd squashfs-root
`
`./AppRun`

limbusdev, 2024-10-21:
One solution is to make a symbolic link from the newer to the older version. There seem to be no breaking changes:

```sh
sudo ln -sf /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37

sudo ln -sf /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.1.so.0 /usr/lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18
```

chen78, 2024-11-13:
> Note that you cannot even run `sudo apt install libwebkit2gtk-4.0-dev` to resolve, because that package no longer exists in Ubuntu 24.04. They've upgraded it to `libwebkit2gtk-4.1-dev`.
> 
> I don't know much about AppImage, but my general gut feeling is that a flatpak wouldn't have this same issue.

that package does not exist

john-parton, 2024-11-13:
It probably depends on your specific version of Ubuntu. I believe there's a Ubuntu-specific AppImage. I understand the irony in having a system-specific version of a file that's supposed to be system-agnostic, but that's what I've been running and it works in my case.

thecameluk, 2024-12-13:
Just install libfuse2t64 , Worked for me!
```shell
sudo apt install libfuse2t64
```