OrcaSlicer/OrcaSlicer #5861: Cooling Overhang Threshold does nothing unless set to 0%

ChaosBlades, 2024-06-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Linux

### OS Version

Pop!_OS jammy 22.04 x86_64

### Additional system information

N/A

### Printer

EnderWire

### How to reproduce

Cooling Overhang Threshold 10-95% all seem to be about 70-75%. 0% does however work.

### Actual results

0%
[image]
10%
[image]
95%
[image]

### Expected results

Overhangs should have increased fan speed based on the degree of overhang from this setting.

### Project file & Debug log uploads

[Cooling Overhang Threshold 3mf.zip](https://github.com/user-attachments/files/15975912/Cooling.Overhang.Threshold.3mf.zip)
[log.zip](https://github.com/user-attachments/files/15975913/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

kt0mmi, 2024-09-04:
I also have the the same issue. The only way to have a different outer wall fan speed is to set the overhang threshold to 0%. Otherwise, all other settings will be ignored, eg. if I set the threshold to 50%, all my overhangs >50% will have base Fan Speed and not the correctly updated one.

otede, 2024-09-04:
@ChaosBlades Is this reproducible in the latest nightly build? https://github.com/SoftFever/OrcaSlicer/releases

ChaosBlades, 2024-09-05:
@otede Yes, exactly the same, no change.

HendrawanAdiWijaya, 2024-10-26:
[image]
I have an object with 45% degree of overhang all the way from the bottom to the top and i set the cooling overhead threshold to 50%, turn out some part has 100% fan speed which is the fan speed for overhangs. Is this a bug or am I just not understanding it?

otede, 2024-10-28:
Any updates on that issue?

dskinder, 2024-11-04:
Any updates? The issue seems quite severe.

otede, 2024-11-06:
Does this affect the built-in temp tower?

galaxyman7, 2024-11-06:
> Does this affect the built-in temp tower?

Yep, here is the sliced file showing fan speed. I have it set to 10% overhang and 100% fan speed. It only detects the overhang some of the time.
This is using 0.1mm layers and 0.56mm line width. 
[image]

[image]

dskinder, 2024-11-30:
Any updates on that issue?

chris99336, 2024-12-13:
I have an update to add to this. I finally found a workaround that makes the fan speed actually be respected for whatever threshold angle you pick. I have been fighting this for a while. Turn on slowdown for overhangs in the print settings even if you do not want it to slow down. Then just set the thresholds that you want to be full speed to 99% speed and suddenly the overhangs are detected and fed into the fan speed setting in the filament.

otede, 2024-12-27:
> @shyblower is working on an issue similar to this, but with the overhang speeds. It seems that overhang detection in general is a little broken right now, including both print speeds and fan speeds, especially at lower overhang percentages (10-25% range), which unfortunately covers most overhangs when you are using lower layer heights like 0.1mm.
> 
> [#7685 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/7685#issuecomment-2539603850)

If you're working on it then it is safe to assume the very recent builds are still not free of that issue.

shyblower, 2024-12-27:
I'm not an official Orca developer, I've just forked Orca and fix bugs and add features I need to my own fork.

I've already taken a brief look into the issue but it doesn't seem to be too easy to fix because there are a lot of interdependent features and code from lots of different programmers, which is sometimes difficult to follow.

I can't guarantee anything, but because it's something that also bothers me I'm trying to fix this issue.
I can't give an ETA but should I manage to make it work I'll post a PR here, which may or may not get included into the main branch by the official OrcaSlicer team.

otede, 2025-01-01:
> I'm not an official Orca developer, I've just forked Orca and fix bugs and add features I need to my own fork.
>
> I've already taken a brief look into the issue but it doesn't seem to be too easy to [...]

I just pointed that out to anyone reading this - there is no point in switching to weekly builds as the Orca team hasn't even begun to tackle the problem. 

Your fork? You've got an Orca fork going on? I haven't seen any such projects on your github profiles.

shyblower, 2025-01-01:
@otede https://github.com/shyblower/OrcaSlicer

v4Lo, 2025-01-13:
It starts being buggy at 0.13mm for me, with tiny spots. 0.12 is full blown broken. 0.14 no problems.

[image]
[image]
[image]

Same spot, same file, 0.12, 0.13, 0.14 top to bottom. This bug makes ASA/ABS overhangs unprintable. Overhang in question is a circle at 45°

closed 2025-04-21 by github-actions[bot]

mattk926, 2025-05-08:
Just recently switched to orcaslicer and am running into this issue on latest 2.3.0 release this still appears to be a problem. This has been driving me mad trying to figure out why these 45degree overhangs on gridfinity bins weren't triggering my overhang fan setting. Any updates from the team on when this will be fixed or if this has worked in a previous version which version that is?

chris99336, 2025-05-10:
> > I have an update to add to this. I finally found a workaround that makes the fan speed actually be respected for whatever threshold angle you pick. I have been fighting this for a while. Turn on slowdown for overhangs in the print settings even if you do not want it to slow down. Then just set the thresholds that you want to be full speed to 99% speed and suddenly the overhangs are detected and fed into the fan speed setting in the filament.
> 
> Since you went to length with trialing this I assume you checked the latest nightly and confirmed the issue is still present?

I'm not a developer and don't go around testing stuff like that. I just shared what I managed to get to work for me and once it started working I just moved on.

chris99336, 2025-05-10:
> Just recently switched to orcaslicer and am running into this issue on latest 2.3.0 release this still appears to be a problem. This has been driving me mad trying to figure out why these 45degree overhangs on gridfinity bins weren't triggering my overhang fan setting. Any updates from the team on when this will be fixed or if this has worked in a previous version which version that is?
> 
> the workaround here [#5861 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/5861#issuecomment-2540387669) seems to work for anyone else that finds this thread so clearly a bug that needs to be fixed.

I am glad people are finding value in this. I was glad when I found it and I do wish the bug would be fixed, but I haven't paid it any mind since this seems to consistently work for me.

SparrowTheNerd, 2026-08-15:
I am running on 2.4.2 and still encountering the same issue. 0% threshold is the only thing that alters the outer perimeter fan speed, and I really don't want 100% cooling on all outer perimeters.