OrcaSlicer/OrcaSlicer #5865: Allow Negative "Exclude Bed Area" values

idontknowtom, 2024-06-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

### Background on Printer / OrcaSlicer Setup

I currently have an older Intamsys Funmat HT printer that uses the stock mechanical system with all new electronics to run Klipper. I have also added a powder coated PEI magnetic bed and dockable probe / Voron Sexbolt Z endstop to the right of the bed.

Homing X and Y uses stock physical endstop switches. These switch postions as well as the length of the physical Y axis mean that I physically cannot move my printer nozzle in the front 24.5mm of the bed, left 8mm of the bed, nor the back 25.5mm of the bed.

I have measured and modeled my bed in Fusion360 for export as STL to use in OrcaSlicer for the "Model" in Printer Settings, Printable Area, Bed Shape window. I have also created a 270x305mm texture (the size of my bed) with a white box that is 260x255mm wide to visually represent Print Area. This is placed at (8, 24.5) in relation to the front left of the bed. Lastly, I have also set the size of my bed in OrcaSlicer to 270x305 and an origin of 8, 24.5

Visually my buildplate setup in OrcaSlicer represents the physical motionability of my printer in X and Y, as well as the 0,0 and relevant coordinates (min and max positions) in Klipper.

### The Issue

The front 24.5mm, left 8mm and top 25.5mm of my bed are not printable regions. Physically the nozzle cant reach these areas (due to X and Y endstops, and the length of the Y axis). Klipper software aligns with these dimensions and will stop printing and report a "Move out of range" error if any gcode goes into these regions.

Users of the printer can slice an object that will place gcode movements in these unprintable regions and the slicer will not warn them. I would like to use the "Exclude Bed Area" input to define these areas, but with the Origin set at 8, 24.5 the front and left regions are considered negative values and will not be accepted. I would like to define:

> -8x-24.5, 260x-24.5, 260x0, 0x0, 0x255, 260x255, 260x305, -8x305

For my printer these coordinates are: Front Left (Bed), Front Right (Bed), Front Right (Printable Area), Front Left (Printable Area) aka X & Y Home, Back Left (Printable Area), Back Right (Printable Area), Back Right (Bed), Back Left (Bed)

3D Model of my printer bed and hot end. Showing the physical 0,0 position due to endstop / motionability limits
[image]

OrcaSlicer Settings for bed size and origin
[image]

OrcaSlicer build plate with texture and model applied. Texture visually (with red) shows areas that are unprintable
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Exclude Bed Area should take into account the user defined bed size and origin and 

### Describe alternatives you've considered

I have added a texture to visually show the areas that are unprintable. But if a user places an object in this region, or a skirt or support overlaps the region, they are not notified. If this gcode is sent to the printer then as soon as motion is requested to move into one of these regions the print stops and gives a "Move out of range" error.

closed 2024-10-03 by github-actions[bot]

Nosnahc, 2024-10-06:
@idontknowtom 
Set your origin correclty and use the "exclude bed area" feature ;)

idontknowtom, 2024-10-06:
> @idontknowtom Set your origin correclty and use the "exclude bed area" feature ;)

@Nosnahc Not sure how this is helpful. What should I consider my origin / (0, 0) position? The front left corner of the bed? Or the furthest front left corner that I can physically access (the home point for X and Y)...which is at (8, 24.5) in relation to the front left corner of the bed.

Nosnahc, 2024-10-06:
Your origin should be set to 0x0, not 8x24.5.
Then apply the bed exclusion area.

Your 0x0 in Orca should be the same as your 0x0 in reality.

If I put 0x0 in Klipper, my printer goes to a position, it should do the same position in Orcaslicer, because you will get negative coordinates otherwise

wohltat, 2025-09-02:
Orca allows to set different origins and the values in "Excluded bad area" are relative to that. So this is clearly a bug!

I use a flashforge printer and cannot change the origin.
My origin is in the middle of the bed and if i add and excluded area ("10x10, 10x0, 0x0, 0x10") it is also (pretty much) in the middle of the bed. There is no way for me to put the excluded area in 3 of the 4 quadrants with negative coordinates.

There is also no real workaround by changing the settings file directly. 
The file is for my linux system in: 
`~/.config/OrcaSlicer/user/default/machine/Flashforge Adventurer 5M Pro 0.4 Nozzle - myProfile.json`

```
"bed_exclude_area": [
        "-96.5x-95",
        "-96.5x-110",
        "-52.5x-110",
        "-52.5x-95"
    ],
```

The values are accepted and even work as excluded area, but
- as soon as you go to the settings the area is deemed invalid and removed
- the display of the area is broken (see screenshot), should be a rectangle

[image]

[image]

The area should look like this:
[image]

Please reopen the issue.

Carter3DP, 2025-09-22:
Please reopen this issue, Scara printers use 0,0 as the inside of the prox arm. Exclusion zones are impossible to implement because of this.

graycoderep, 2025-10-31:
We need this feature! My origin is x=200 y=175, not 0x0!
At the same time I have left extruder and to prevent collision I need to exclude that area.

[image]