OrcaSlicer/OrcaSlicer #5868: Orca 2.1.0 starts loop in mid air

Rickthebig, 2024-06-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

DIY BLV Cube

### How to reproduce

Slice the model in the attached project file.
[image]

Go to Gcode line 193@48295 and see how Orca tries to extrude beginning in mid air if "Bridge counterbore holes" set to "Partially bridged".

### Actual results

With "Bridge counterbore holes" set to "Partially bridged":
[image]

### Expected results

With "Bridge counterbore holes" set to "None" the layer looks like it should:
[image]

### Project file & Debug log uploads

[Printing_in_mid_air_bridge_counterbore_hole.zip](https://github.com/user-attachments/files/15986670/Printing_in_mid_air_bridge_counterbore_hole.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

A slicer should never start a loop in mid air.

kylek29, 2024-06-26:
I observed this with another model as well.

tlhintoq, 2024-07-01:
You see it with `organic` supports too.  Sometimes it makes a partial second/interior wall of the support just out in the middle of the air.

closed 2024-10-08 by github-actions[bot]