OrcaSlicer/OrcaSlicer #5877: Non-standard nozzle diramater for BL printers support

rednaxal, 2024-06-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm using a non-standard 0.3mm nozzle on my BL P1P as a best compromise between speed, reliability and detailing. 
Up until V2.1.0-Beta I was able to still send a print job to printers. It was showing me a warning but was allowing me to ignore it still. 
Since V2.1.0-RC it no longer offers the ability to ignore the warning and just doesn't allow to send a job for printing in such case. 

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Warn a user that the nozzle diameter in the BL printer accessories config is different that the one, specified on OrcaSlicer's profile but still allow a user to ignore/bypass this warning and send a print job to printer. Or even better provide a setting somewhere to suppress this kind of warning/error message for an experienced users who know what they're doing.

tlhintoq, 2024-07-01:
Wait... Huh?
You're slicing for one nozzle size and want to send to a machine with a different nozzle size than you sliced for?
That sounds like all levels of bad.

A - Just make another printer with the right size nozzle.  
[image]

or

B - Don't worry about the nozzle size for such a petty difference.  Keep the .4 nozzle on the machine and slice using a .3 line width and `Arachne` wall generator.  I bet you won't be able to tell any difference.  You can even set all the other walls to .4 or .5 so the insides and infil go FASTER and keep the exterior wall at .3 for detail.

C - Set the line widths as percentages.  That way the profile works when you select a machine with a .6 nozzle or a .4 nozzle. And you still get the balance of speed to quality possible for that given machine.  
[image]

rednaxal, 2024-07-01:
@tlhintoq 
You totally misunderstood my problem. I'm slicing for a 0.3 nozzle. I know how to configure printer settings. :)
[image]

But when I'm sending this to BL printer on older versions of OrcaSlicer I'm getting this:
[image]
which allows me to still print this. On a latest version I'm not longer able to bypass this warning.

tlhintoq, 2024-07-02:
Duplicate to https://github.com/SoftFever/OrcaSlicer/issues/5849

@rednaxal What OS?  Because on that other report for the same issue there is another person and myself both showing the current version still having the ability to `[CONFIRM]` and continue on anyway.

rednaxal, 2024-07-02:
@tlhintoq 
Windows 10. 
I would assume that you're using a V2.1.0-Beta or earlier. The problem introduced in V2.1.0-RC and later.

Highlander-bdj, 2024-07-20:
I have this same problem on version 2.1.1 for the Mac.  It is NOT present in version 2.0.

Highlander-bdj, 2024-07-30:
This should be labeled as a bug not an enhancement.

anselor, 2024-08-16:
This has been re-fixed in the code but was merged right after the last release so is still unavailable. It looks like there was a new nightly build 2 days ago so it's probably available there.

mrelwood, 2024-08-22:
Thank goodness I found this topic! And thank you anselor, the latest nightly (Aug 8th 2024) indeed has the Confirm button back. I can finally print with a 0.3mm nozzle again!

closed 2024-11-28 by github-actions[bot]