OrcaSlicer/OrcaSlicer #5878: OrcaSlicer crashes when attempting to open the camera feed for Bambulab A1 Mini

ronoaldo, 2024-06-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Debian GNU/Linux 12 (bookworm)

### Additional system information

OS: Debian GNU/Linux 12 (bookworm) x86_64 
Kernel: 6.1.0-22-amd64 
Resolution: 5120x1440 
DE: Plasma 5.27.5 
WM: KWin 
Theme: Breeze [Plasma], Adwaita-dark [GTK2/3] 
Icons: [Plasma], breeze-dark [GTK2/3] 
CPU: AMD Ryzen 9 5900X (24) @ 4.200GHz 
GPU: AMD ATI Radeon RX 6800/6800 XT / 6900 XT 
Memory: 5563MiB / 32000MiB

### Printer

Bambulab A1 

### How to reproduce

1. Install Bambulab Network Plugin
2. Login to your MakerWorld account using the plugin
3. Add the A1 Mini printer (or sync if already added previously)
4. Go to Device tab
5. Notice you can see all other info, including AMS related filaments
6. Click the 'Play' button under the camera feed
7. Orca will crash with a segmentation fault

### Actual results

The application crashes instead of showing the camera feed after the "Loading" message is displayed next to the play button.

### Expected results

The camera feed would be expected to show up instead.

### Project file & Debug log uploads

Attached the logs of the most recent session that caused the crash. There is no need to have a project open to reproduce the crash.

[logs.tar.gz](https://github.com/user-attachments/files/15993377/logs.tar.gz)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

[image]

ronoaldo, 2024-06-26:
The same issue does not happen on BambuStudio, which I'm currently using for both A1 Mini and K1 Max.

oltas123, 2024-07-21:
Same issue but with both OrcaSlicer and Bambu studio on Debian 12. Tryed it only in Lan Mode

oltas123, 2024-07-21:
Same issue but with both OrcaSlicer and Bambu studio on Debian 12. Tryed it only in Lan Mode. But works with Manjaro from USB on the same Computer.

rmeissn, 2024-10-21:
Still an issue with v2.1.1

IetIesAai, 2024-10-28:
> Still an issue with v2.1.1

and 2.2.0, Fedora, cloud mode

3dRikal, 2024-10-29:
Same Arch/Catchy-OS

danexpo, 2024-11-07:
Same issue with A1 Mini, P1S on Fedora 41. 

Issue is fixed in Bambu Slicer's latest release [1.10.0 Public Beta2](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81)

mf81234, 2024-11-07:
Can confirm same issue on Windows 11 with X1C, fixed in Bambu [1.10.0 Public Beta2](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81)

3dRikal, 2024-11-07:
No issues in current release of Bambu Studio

ronoaldo, 2024-11-16:
I managed to reproduce the same behavior on BambuStudio 1.9.7.52 - it crashes when launching the webcam. It looks like the issue is related to a version/release of the encoding/decoding of video, since BambuStudio packages a version of the libraries that usually ship with ffmpeg. Also, the network plugin is redownloaded if you first open and run 1.9.7.52 then run the latest 1.10.1 - I'm feeling that Orca may be running an outdated/incompatible version of the plugin and/or it is trying to use the system ffmpeg libs.

rmeissn, 2025-02-21:
Still an issue with Bambulab 2.3.0-beta on Fedora 41

bseidenberg, 2025-03-24:
I can replicate on 2.2.0 on Debian. 

Running it under GDB, this is what the fault and stacktrace look like:

```
Thread 192 "source:src" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffef55fb6c0 (LWP 19598)]
____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
167	../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S: No such file or directory.
(gdb) bt
#0  ____longjmp_chk ()
    at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
#1  0xa4392d52d2fe981c in  ()
```

mf81234, 2025-03-24:
> I can replicate on 2.2.0 on Debian.
>
> Running it under GDB, this is what the fault and stacktrace look like:
>
> ```
> Thread 192 "source:src" received signal SIGSEGV, Segmentation fault.
> [Switching to [...]

Are you able to test with 2.3.0? For me (Windows 11) somewhere between beta2 and rc fixed this issue

bseidenberg, 2025-03-25:
Yes, still crashes on 2.3.0:

```
Thread 180 "source:src" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffef2ffd6c0 (LWP 84007)]
____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
167	../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S: No such file or directory.
(gdb) bt
#0  ____longjmp_chk ()
    at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
#1  0x772b1f5d3028c8ec in  ()
(gdb) 
```

bseidenberg, 2025-03-25:
This is the AppImage (not the Ubuntu-22.04 version) from the GH releases page.

bendwv, 2025-05-11:
Still getting this issue on Fedora 42 kernel 6.14.5-300.fc42.x86_64 (64-bit) with KDE Plasma. Using the appimage.

nyghtly-derek, 2025-11-17:
Encountering this problem using OrcaSlicer 2.3.1 on Bazzite (`bazzite-dx-nvidia:stable`) with my Bambu Lab P1S.

Video works fine in Bambu Studio.

Before I began seeing segfaults, I had a separate issue with missing H264 codecs. Fixed that by installing some gstreamer plugins.