OrcaSlicer/OrcaSlicer #5882: Immediate Crash cant even open v2.1.1 when trying to Mac

JayDubYa14, 2024-06-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

macOS 11.7.8 (20G1351)

### Additional system information

Model: iMac15,1, BootROM 478.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 4 GHz, 32 GB, SMC 2.23f11
Graphics: kHW_AMDRadeonR9M295XItem, AMD Radeon R9 M295X, spdisplays_pcie_device, 4 GB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 0/DIMM1, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM1, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x142), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1680.11)
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE SSD SM1024F, 1 TB
USB Device: USB 3.0 Bus
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
Thunderbolt Bus: iMac, Apple Inc., 26.1

### Printer

Cant even open program

### How to reproduce

try to open application

### Actual results

immediate crash

### Expected results

should open program

### Project file & Debug log uploads

not sure what this means

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

version 2.1.0 works fine, just tried updating and cannot open new version. Tried uninstalling everything, reset computer, reinstalling, version 2.1.1, crashes right away.

gixxerfool, 2024-06-26:
I have this same issue. I cant find an older version to roll back to either.

JayDubYa14, 2024-06-26:
> I have this same issue. I cant find an older version to roll back to either.

https://github.com/SoftFever/OrcaSlicer/releases/download/v2.1.0/OrcaSlicer_Mac_x86_64_V2.1.0.dmg

gixxerfool, 2024-06-26:
> > I have this same issue. I cant find an older version to roll back to either.
> 
> https://github.com/SoftFever/OrcaSlicer/releases/download/v2.1.0/OrcaSlicer_Mac_x86_64_V2.1.0.dmg

Thanks.  Missed the assets arrow.

SoftFever, 2024-06-27:
@JayDubYa14 Thank you for reporting the issue.
Seems like an issue for Intel Mac machine only.

@gixxerfool could you confirm is your Mac is also an Intel CPU machine?

gixxerfool, 2024-06-27:
> @JayDubYa14 Thank you for reporting the issue. Seems like an issue for Intel Mac machine only.
> 
> @gixxerfool could you confirm is your Mac is also an Intel CPU machine?

That’s an affirmative.  I rolled back to 2.0.0 for now.

SoftFever, 2024-06-27:
> > @JayDubYa14 Thank you for reporting the issue. Seems like an issue for Intel Mac machine only.
> > @gixxerfool could you confirm is your Mac is also an Intel CPU machine?
> 
> That’s an affirmative. I rolled back to 2.0.0 for now.

Also OS X 11?

JayDubYa14, 2024-06-27:
Yes, Intel iMac i7 quad-core processor, iOS 11.7.8

gixxerfool, 2024-06-27:
> > > @JayDubYa14 Thank you for reporting the issue. Seems like an issue for Intel Mac machine only.
> > > @gixxerfool could you confirm is your Mac is also an Intel CPU machine?
> > 
> > 
> > That’s an affirmative. I rolled back to 2.0.0 for now.
> 
> Also OS X 11?

11.7.10

atl285, 2024-06-27:
The same here... macOS 11.7.10 on Intel Mac mini 2012 (i7-3720QM) and Orca Slicer 2.1.1 is crashing on launch.

naidoff, 2024-06-27:
Also cras, 11.7.10

Hajo-kama, 2024-06-28:
I have also this problem intel Macbook pro with OS 11.7

mclardass, 2024-06-29:
Same, Intel MBP running Big Sur. Orca 2.1.0 works but 2.1.1 will not open.

cnorrick, 2024-06-29:
I also have this problem. Intel MacBook Pro, MacOS 11.7.10. 2.8 GHz Dual-Core Intel Core i7

jorgelserve, 2024-07-01:
Hi everyone,

I'm also experiencing the **EXC_BAD_ACCESS (SIGSEGV)** crash on the main thread as reported by others. This issue only occurs in version 2.1.1 of OrcaSlicer, but not in version 2.1.0. I hope this information is useful @SoftFever 

    Process:               OrcaSlicer [21069]
    Path:                  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
    Identifier:            com.softfever3d.orca-slicer
    Version:               2.1.1 (???)
    Code Type:             X86-64 (Native)
    Parent Process:        ??? [1]
    Responsible:           OrcaSlicer [21069]
    User ID:               501
    
    Date/Time:             2024-07-01 17:32:35.216 -0500
    OS Version:            Mac OS X 10.15.7 (19H2026)
    Report Version:        12
    Anonymous UUID:        A0A68FA2-9C08-CF04-CDC7-ADDE4C136E97
    
    Sleep/Wake UUID:       3EA50DC7-35D1-4C3F-A2E5-4B9B65747E94
    
    Time Awake Since Boot: 130000 seconds
    Time Since Wake:       12000 seconds
    
    System Integrity Protection: enabled
    
    Crashed Thread:        0  Dispatch queue: com.apple.main-thread
    
    Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
    Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000004d29000
    Exception Note:        EXC_CORPSE_NOTIFY
    
    Termination Signal:    Segmentation fault: 11
    Termination Reason:    Namespace SIGNAL, Code 0xb
    Terminating Process:   exc handler [21069]
    
    VM Regions Near 0x4d29000:
    --> 
        __TEXT                 0000000104d29000-0000000109381000 [ 70.3M] r-x/r-x SM=COW  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
    
    Application Specific Information:
    /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
    
    Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
    0   com.softfever3d.orca-slicer   	0x0000000104ddfb9e __cxx_global_var_init.609 + 14
    1   dyld                          	0x000000010f025353 ImageLoaderMachO::doModInitFunctions(ImageLoader::LinkContext const&) + 535
    2   dyld                          	0x000000010f02575e ImageLoaderMachO::doInitialization(ImageLoader::LinkContext const&) + 40
    3   dyld                          	0x000000010f02017b ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 493
    4   dyld                          	0x000000010f01e234 ImageLoader::processInitializers(ImageLoader::LinkContext const&, unsigned int, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 188
    5   dyld                          	0x000000010f01e2d4 ImageLoader::runInitializers(ImageLoader::LinkContext const&, ImageLoader::InitializerTimingList&) + 82
    6   dyld                          	0x000000010f00c6a8 dyld::initializeMainExecutable() + 199
    7   dyld                          	0x000000010f011bf7 dyld::_main(macho_header const*, unsigned long, int, char const**, char const**, char const**, unsigned long*) + 6728
    8   dyld                          	0x000000010f00b227 dyldbootstrap::start(dyld3::MachOLoaded const*, int, char const**, dyld3::MachOLoaded const*, unsigned long*) + 453
    9   dyld                          	0x000000010f00b025 _dyld_start + 37
    
    Thread 1:
    0   libsystem_pthread.dylib       	0x00007fff6cb5ab68 start_wqthread + 0
    
    Thread 2:
    0   libsystem_pthread.dylib       	0x00007fff6cb5ab68 start_wqthread + 0
    
    Thread 0 crashed with X86 Thread State (64-bit):
      rax: 0x000000010f0a2268  rbx: 0x0000000104ddfb90  rcx: 0x00007ffeeaed6c78  rdx: 0x00007ffeeaed6c20
      rdi: 0x0000000000000001  rsi: 0x00007ffeeaed6c10  rbp: 0x00007ffeeaed26d0  rsp: 0x00007ffeeaed26c0
       r8: 0x000000010f0a2330   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
      r12: 0x0000000000000000  r13: 0x00007ffeeaed26e0  r14: 0x0000000004d29000  r15: 0x0000000000000000
      rip: 0x0000000104ddfb9e  rfl: 0x0000000000010202  cr2: 0x0000000004d29000
      
    Logical CPU:     0
    Error Code:      0x00000004 (no mapping for user data read)
    Trap Number:     14
[... 331 lines cut ...]
    __FONT_DATA                          4K        1 
    __LINKEDIT                       422.3M        4 
    __OBJC_RO                         32.3M        1 
    __OBJC_RW                         1908K        2 
    __TEXT                           344.2M      297 
    __UNICODE                          564K        1 
    mapped file                       29.3M        2 
    shared memory                       48K        6 
    ===========                     =======  ======= 
    TOTAL                              1.4G      650 
    TOTAL, minus reserved VM space     1.1G      650 
    
    Model: iMac14,1, BootROM 433.140.2.0.0, 4 processors, Quad-Core Intel Core i5, 2,7 GHz, 8 GB, SMC 2.14f24
    Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
    Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343531533641465238412D50422020
    Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343531533641465238412D50422020
    AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x111), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1615.7)
    Bluetooth: Version 7.0.6f8, 3 services, 27 devices, 1 incoming serial ports
    Network Service: Wi-Fi, AirPort, en1
    Serial ATA Device: CT960BX500SSD1, 960,2 GB
    USB Device: USB 3.0 Bus
    USB Device: BRCM20702 Hub
    USB Device: Bluetooth USB Host Controller
    USB Device: FaceTime HD Camera (Built-in)
    Thunderbolt Bus: iMac, Apple Inc., 23.10

Rekiem24, 2024-07-03:
Hi!
Same issue here.
[image]
Bigsur 11.7.10

My3DBr, 2024-07-03:
Same with me, Mac OS Catalina v10.15.7
Core i5

dgudtsov, 2024-07-03:
The same issue with 2.1.1 on Intel Mac running BigSur (latest). No issue with 2.1.0 on the same Mac.
No issue with 2.1.1 on Intel Mac running latest Ventura.
Please check what's wrong with build 2.1.1.

fludstank, 2024-07-13:
Per [this post](https://github.com/SoftFever/OrcaSlicer/pull/6002#issuecomment-2211556307) @Noisyfox was able to get it fixed - I'm going to test it out as well and hope this can help others that may not have seen the fix.

RockChewer3D, 2024-08-28:
I had this fixed by the Apple firewall must have updated and crashed 2.1.1 again. Here is what I did to make everything work, and hopefully it will stick.

1. Delete all copies off your computer in Applications and the Downloads
2. Download version 2.0.0, it will work with Big Sur 11.7, if you have something else you may need an earlier version
4. Open it and move OrcaSlicer to Applications
5. Download version 2.1.1, open it - make sure you add it as a second version
6. Go to System Preferences - Security & Privacy - Firewall
7. Unlock the lock in the lower left with you password
8. Click on Firewall Options
9. Click the "+" sign to add
10. Find "OrcaSlicer 2" or whatever you named the second version, then make sure it is a green dot saying "Allow Incoming Connections"
11. Click OK
12. Click the Lock to lock it again
13. Exit Security and Privacy - System Preferences
14. Open "OrcaSlicer 2" and it should run just fine.
[image]

Noisyfox, 2024-08-28:
> I had this fixed but the Apple firewall must have updated and crashed 2.1.1 again. Here is what I did to make everything work, and hopefully it will stick.
> 
> 1. Delete all copies off your computer in Applications and the Downloads
> 2. Download version 2.0.0, it will work with Big Sur 11.7, if you have something else you may need an earlier version
> 3. Open it and move OrcaSlicer to Applications
> 4. Download version 2.1.1, open it - make sure you add it as a second version
> 5. Go to System Preferences - Security & Privacy - Firewall
> 6. Unlock the lock in the lower left with you password
> 7. Click on Firewall Options
> 8. Click the "+" sign to add
> 9. Find "OrcaSlicer 2" or whatever you named the second version, then make sure it is a green dot saying "Allow Incoming Connections"
> 10. Click OK
> 11. Click the Lock to lock it again
> 12. Exit Security and Privacy - System Preferences
> 13. Open "OrcaSlicer 2" and it should run just fine.
>     ![image](https://private-user-images.githubusercontent.com/174746110/362115552-684ee46a-d504-40f9-88b1-5e51c9fbc7e5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ4MjU5MjQsIm5iZiI6MTcyNDgyNTYyNCwicGF0aCI6Ii8xNzQ3NDYxMTAvMzYyMTE1NTUyLTY4NGVlNDZhLWQ1MDQtNDBmOS04OGIxLTVlNTFjOWZiYzdlNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyOFQwNjEzNDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYTdhZGY2NGUxNDc5MDBmY2E3NzRiNzJiZDExZGI1ZTAzZjc3MjFlYjMyNmRiMzJkOWU4MTNkNzkyY2Y3OTBmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5JgqGACIRRqcXtzB5L1qZ9oOUWep4iZD3V508XzKdW8)

The latest nightly build should have this issue fixed I believe. Could you give it a try and see if it works?

RockChewer3D, 2024-08-29:
Hi @Noisyfox , YES! The 2.2.0-dev worked brilliantly! Thank you so much. As a community we cannot thank you enough, the team is amazing and Orca is far superior!

closed 2025-01-25 by github-actions[bot]