OrcaSlicer/OrcaSlicer #5883: Print bed model seems to bed offset to -x and -y for a small amount.

subdancer, 2024-06-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There seems to be a small issue with the print bed model position. It's slightly offset to -x and -y while using windows. I just noticed it when i changed it and zoomed in. The image texture seems to be correct as we can see on the arrows of the origin.
The origin of my bed model stl itself is perfectly centered. Here is an image that shows the offset.
[image]

To double check if it's just me i noticed the offset is also visible on a picture from release 2.1.1 that has been released today. 
see [here](https://private-user-images.githubusercontent.com/1537155/342066561-57111aa6-992c-4989-b3b3-1832411f5774.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk0Mjc4NDgsIm5iZiI6MTcxOTQyNzU0OCwicGF0aCI6Ii8xNTM3MTU1LzM0MjA2NjU2MS01NzExMWFhNi05OTJjLTQ5ODktYjNiMy0xODMyNDExZjU3NzQuZ2lmP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MjZUMTg0NTQ4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDZhYzBmMTljNDczY2M5NGFiOWRhMWE0MDk3YzhmNTlmZjA5YzE0NzRkMjZkNTM5MDE5MzVkMDhkMGNkNDM4ZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.-zCY8Wv09UcV0XtbL5OonQH358xy_hqrdBhcMG8FuOA) .  The left border of the print bed model on x to origin (0,0,0) is bigger than on the right border/side of the bed, same goes for y axis. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It's only a small visual bug/issue that does not effect slicing or printing in any way but it would be nice if things are perfectly aligned.

subdancer, 2024-06-28:
i just tried it in prusaslicer to just check if i did something wrong in the model but prusaslicer renders it nicely as seen here 
[image] . It seems there are differences in the slic3r implementation. And yes. I'd say this definitely needs a cleanup. Can someone pinpoint the offset?

Orca:
https://github.com/SoftFever/OrcaSlicer/blob/31df3f01ea70d61e9dcc7680977feb10e5612a40/src/slic3r/GUI/3DBed.cpp#L668

Prusa:
https://github.com/prusa3d/PrusaSlicer/blob/fd02400245df25f1286059aa642739660df76663/src/slic3r/GUI/3DBed.cpp#L424

kylek29, 2024-06-28:
See's it's also on the Z-axis, as the bed plate is the same distance below, see comparison screenshot below (PrusaSlicer top, OrcaSlicer bottom, view difference is because Prusa lacks the Ortho-view modes). Orange line on the zoomed in part is the true-Z level.

[image]

subdancer, 2024-06-29:
yes, you are right. Z is also slightly offset. 
[image] 

Guess that's that  -0.41 in Line 625. I don't understand the code completly but hardcoded values and changing dimensions of models don't go well together. I guess it's back from the days when models were fixed, not sure though. https://github.com/SoftFever/OrcaSlicer/blob/31df3f01ea70d61e9dcc7680977feb10e5612a40/src/slic3r/GUI/3DBed.cpp#L625

ilko-k, 2024-09-15:
same here, model and texture are misaligned, but no issue with prusaslicer

Orca :
[image]

Prusa :
[image]

subdancer, 2024-09-18:
Yes, it's the same issue. The texture position seems to be fine. I guess it's just the bed model position that is not aligned properly to the orgin of the print volume. Must be a leftover from an old codebase. My guess is it happend when slic3r got changeable bed models.  It does not effect printing or slicing in any way but it would still be nice to get this fixed, because it's not nice to have inaccurate visuals. The hardcoded z-offset tells me that this belonged to a different scenario. I might be wrong though.

closed 2024-12-26 by github-actions[bot]