OrcaSlicer/OrcaSlicer #5884: 2.1.1 Crashes when saving filament or creating filament profiles

justinr54321, 2024-06-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Processor	13th Gen Intel(R) Core(TM) i7-13700HX   2.10 GHz
Installed RAM	32.0 GB (29.7 GB usable)
1GB SSD
GEFORCE RTX 4060 GPU

### Printer

Merc One Plus

### How to reproduce

Open or create a filament profile
Make changes
Click on save

### Actual results

Orca 2.1.1 freezes and crashes

### Expected results

Profile saved and able to move on

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15995154/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

SoftFever, 2024-06-27:
Can you profile more info like how to reproduce it step by step?
I can't reproduce it

justinr54321, 2024-06-27:
I do apologize, meant to close this ticket, as it started to work about an hour after I posted this. But when I did was made a change to filament color and flow rate and then save it,  it would freeze and crash. When I tried to do a new filament profile and save it, same thing. Gotta love software sometimes. Lol

justinr54321, 2024-06-27:
Closing issue

Noisyfox, 2024-06-27:
Could you send me your entire %AppData%\OrcaSlicer folder?

justinr54321, 2024-06-27:
[OrcaSlicer.zip](https://github.com/user-attachments/files/16007878/OrcaSlicer.zip)
[OrcaSlicer.zip](https://github.com/user-attachments/files/16007882/OrcaSlicer.zip)

Noisyfox, 2024-06-27:
Could you tell me which profile you have that could trigger this crash?

justinr54321, 2024-06-27:
> Could you tell me which profile you have that could trigger this crash?

It was the COEX Metallic Silver ABS

Noisyfox, 2024-06-27:
Does the same issue occur on 2.1.0?

justinr54321, 2024-06-27:
It didn't. Happened after I updated, almost as if it had difficulties
communicating with the directory that held the filament profiles.

closed 2024-10-03 by github-actions[bot]