OrcaSlicer/OrcaSlicer #5885: variable layerheight crash to desktop on Linux Mint

Chimorinxx, 2024-06-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.3 Cinnamon

### Additional system information

Macbook Pro 2013 13"
CPU: Intel© Core™ i5-3210M

### Printer

Bambu Lab X1-C

### How to reproduce

Click model in viewer
click icon for variable layerheight

### Actual results

crash to desktop

### Expected results

not crash to desktop i guess

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15995298/log.zip)
[project_file.zip](https://github.com/user-attachments/files/15995299/project_file.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

i run the provided AppImage as executable

Noisyfox, 2024-06-27:
Do you have system crash log?

mtexier, 2024-07-08:
Hi, I have the same issue on Suse Linux.
How do I get the system crash log.
From the command line it only says "segmentation fault".

AMiketta, 2024-09-09:
hi, i have the same issue in elementary OS

Noisyfox, 2024-09-09:
Depends on the distro the log path can be different so it's hard to tell. Maybe try location like ' /var/log/syslog' etc.

origin2000, 2024-09-22:
**OrcaSlicer Version:** Stable 2.1.1, Nightly Build Sun 9/22 and earlier 2.2.0 beta, QidiStudio 1.9.5.51
**Printer:** Qidi Q1 Pro
**OS Version:** openSUSE Leap 15.5 - Wayland Graphics Platform NOT X11

This happened to a Linux Mint 22 (based upon Ubuntu 2404) as well with the exception of Nightly Build  2.2.0 beta, which is why I moved to 2.2.0 beta early on.

I can support testing on openSUSE Leap 15.5. By mid Oct. I'll also have access to the Mint 22 system again.

origin2000, 2024-09-22:
Gave BBL studio a quick trial: 
openSUSE Leap 15.5
- fedora-v01.09.07.52, does not start at all, reason: GLIBC_2.32 or newer required, openSUSE 15.5 is GLIBC_2.31.
- ubuntu-v01.09.07.52-24.04, does not start at all, reason: GLIBC_2.32 or newer required, openSUSE 15.5 is GLIBC_2.31
- ubuntu-v01.09.07.52-20.04, starts, works well BUT shows same bug when selecting variable layerheight

Noisyfox, 2024-09-23:
> **OrcaSlicer Version:** Stable 2.1.1, Nightly Build Sun 9/22 and earlier 2.2.0 beta, QidiStudio 1.9.5.51 **Printer:** Qidi Q1 Pro **OS Version:** openSUSE Leap 15.5 - Wayland Graphics Platform NOT X11
> 
> This happened to a Linux Mint 22 (based upon Ubuntu 2404) as well with the exception of Nightly Build 2.2.0 beta, which is why I moved to 2.2.0 beta early on.
> 
> I can support testing on openSUSE Leap 15.5. By mid Oct. I'll also have access to the Mint 22 system again.
> 
> P.S.: Finally decided to create a github account to be able to comment as it very much looks like a platform issue given that it happens on stable, beta and even QidiStudio. May try Bambu, too

Could you take a look at places like `/var/log`, `/var/log/crashes/` etc and see if you can find the crash report of Orca? Thanks.

Noisyfox, 2024-09-23:
> This is a core dump from openSUSE Leap 15.5, OrcaSlicer_Linux_2.2.0-beta.AppImage. I couldn't upload even though I zipped, as the limit is 25MB. Here is a public link to my MEGADrive: https://mega.nz/file/sY5FWCZD#4sKIm-OX9xIIVxNxxOUn7aMOzR9ruPzogPrc0iZpL8c The archive is 57.4 MB. Pls let me know whether you could download successfully. I also can install Mint22 (ubuntu flavour) in a VirtualBox VM and test there too if need be. Cheers.

Thanks. I downloaded it successfully and will take a look soon.

origin2000, 2024-09-24:
Hi Noisyfox, spent the rest of the day testing and investigating the core dump. The reason for the core dump is a change in used library versions that are not compatible with openSUSE 15.5 any longer. I also recognized that 15.5 has been updated to 15.6 and support for 15.5. will end 12/24.  So I upgraded to 15.6.

- OrcaSlicer_Linux_V2.1.1.AppImage, 
- OrcaSlicer_Linux_V2.2.0-beta.AppImage, 
- QIDIStudio_1.9.5.51_Linux.AppImage

seem to work now.  For sure selecting variable layer-height does not crash the app any longer. 
I bet this is true for Ubuntu 2404 LTS  flavors as well although I didn't test. People still on an Ubuntu 2004 LTS flavor therefore should upgrade to their respective 2404 editions.  

This being said the solution to the bug (incompatible libs)  seems to be to  upgrade to your latest stable Linux distribution.

If I'd be you I'd close the bug.  Thank you so much for your support!

Chimorinxx, 2024-09-24:
[syslog.zip](https://github.com/user-attachments/files/17111167/syslog.zip)
Here is the systemlog. Forgot about it. fresh system boot straight to orcaslicer

origin2000, 2024-09-24:
@Chimorinxx: Are you still on Mint 21.3? It's based upon glibc 2.31 whereas the newer Orca's seem to require newer versions. Can you upgrade to Mint 22?

Noisyfox, 2024-09-24:
> This is a core dump from openSUSE Leap 15.5, OrcaSlicer_Linux_2.2.0-beta.AppImage. I couldn't upload even though I zipped, as the limit is 25MB. Here is a public link to my MEGADrive: [...]

Btw, which binary were you running when generating this core dump? The ubuntu2004 or 2404?

Chimorinxx, 2024-09-24:
@origin2000 thanks for pointing this out. Didnt even see there was a new version available. Upgrade to it did solve the issue.

origin2000, 2024-09-24:
@Noisyfox the core dump you received was 2404. But even the 2004 did seg fault.. I believe (not know for sure) that 2.1 was the last version not dumping. With the release of 2.1.1 it started. I was only providing the 2404 dump cause I thought focusing on 2.2 makes more sense ...  But also look at @Chimorinxx comment. After upgrading to Mint 22 (ubuntu flavor 2404) the issue went away. It would make sense to further investigate to figure out what specific libraries are affected. This way one could document  in the release notes of future Orca 2.2 that people need to upgrade their systems.

closed 2025-01-01 by github-actions[bot]