OrcaSlicer/OrcaSlicer #5891: impossible choose Filament type for Flying bear Ghost 6

gavrik, 2024-06-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

Sonia 14.5

### Additional system information

Apple M3 Max 16Gb Ram

### Printer

Flying Bear Ghost 6

### How to reproduce

1. Install 2.1.1 version
2. find the Ghost 6 printer profile
3. On the "Filament Selection" you will see nothing to choose

### Actual results

on a screen you will see nothing to select. The Next button will open Error popup which say you should check one of the filament type. 
it is impossible skip this step.
[image]

### Expected results

the  list of the checkbox with filament types.

### Project file & Debug log uploads

No project created yet

### Checklist of files to include

- [ ] Log file
- [ ] Project file

axeok, 2024-07-01:
I faced the same problem.

For "skip" filament selection just go back and select additional printer(for example: klipper 0.4mm).
Then proceeed filament selection and after save - select printer and filament in "Prepare" window.

[image]

gavrik, 2024-07-01:
Confirm. this worked for me.

closed 2024-10-08 by github-actions[bot]