OrcaSlicer/OrcaSlicer #5893: New Filament Creation no longer working

3D-Schmied, 2024-06-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Win 11 Pro

### Printer

Elegoo Neptune 4 Plus

### How to reproduce

1. Go to Filament Section
2. Click on "Filaments to use"
3. Click on "Custom Filaments"
4. Click on "Create New"
5. Fill in Vendor, Type, Serial
6. Choose Filament which should the new one being based on
7. Click on Create

### Actual results

Nothing is happening as you can't select a printer the filament should be use for. No Window with like in  former Versions is popping up. Also installed priners  are  not shown  on the "create  filament" Window.

### Expected results

Selection of at least one installed printer should be possible

### Project file & Debug log uploads

No Log File or other. Just a screenshot available

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

Opepo1, 2024-07-03:
I had the same problem, I went back to version 1.9

GGianpaolo, 2024-08-12:
have the same problem, went back to 2.0.0, so it works again...

andrey-helldar, 2024-09-14:
2.2.0 Beta version too :(

[image]
[image]

andrey-helldar, 2024-12-14:
In the official 2.2.0 release, the problem also persists.

[image]
[image]

closed 2025-03-23 by github-actions[bot]