OrcaSlicer/OrcaSlicer #5896: "Retract amount before wipe" setting isn't working causing severe stringing

Redhorde, 2024-06-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

QIDI Q1 Pro

### How to reproduce

1. Set "Retract amount before wipe" in printer extruder settings to 0%
2. Slice file

### Actual results

After inspecting the Gcode, retraction before wipe still happens and takes away most of the overall retraction despite being set to 0%
[image]

The values of retractions during wipe are much smaller than expected

### Expected results

Here's a picture of the same gcode fragment in QIDI slicer using the exact same retraction settings. Notice how no retraction happens at the start of the wipe and retraction values during wipe are much greater.
[image]

Here's a picture showing how much of a difference it makes in PETG printing quality, using the same printer settings, material settings and process settings.
[image]

### Project file & Debug log uploads

[Retraction test.zip](https://github.com/user-attachments/files/16014684/Retraction.test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2024-06-27:
Orca slicer respects your retraction speed during wiping. If you try to retract while wiping over a very small area, like the retraction test above, at a fast print speed, and combined with a slow retraction speed you are likely to not have enough space to perform the full retraction over that distance.

So Orca will calculate how much filament it can retract while wiping while also respecting how fast you have told it to retract. Then any retraction amount that is left, will be retracted before the wipe to minimise stringing.

If you want to retract more while wiping you have 2 options:
1. Increase your retraction speed. Set this in the extruder settings. Not recommended though as you dont want to retract too fast and cause your filament to separate from the extruder.
2. Reduce your wipe speed (the setting is highlighted in orange below)
3. Increase wipe distance

[image]

The Qidi slicer retracts in your example above 0.175mm of filament over a travel move of 0.318mm performed at 400mm/sec. **This means it is retracting your filament at a speed of 220.13mm/sec** (line 13981 in your picture above).

From experience this can cause filament grinding, extruder stalls and / or extruder skipping. 

Orca on the other hand respects your retraction speed so it retracts slower while also retracting the left overs prior to the wipe move. If you want the full retraction to happen while wiping, increase your wipe distance, reduce your wipe speed and increase the retraction speed.

Redhorde, 2024-06-27:
@igiannakas  Thank you for your explanation, it seems that "Retract amount before wipe" does in fact work as intended.
Before I close this issue, could you tell me one more thing before I open a second one.
"Role base wipe speed" should force wiping speed to be set to the last performed line role shouldn't it? In that case why doesn't it inherit the 20mm/s that my outer wall prints at, instead being set to 159mm/s?

igiannakas, 2024-06-27:
I think it’s because your part is being slowed down due to layer cooling. Try disabling it in the filament settings and check whether the speed remains the same.

If it does, this may be a bug ;)

Redhorde, 2024-06-27:
Right, my outer walls don't usually print at 20mm/s so it's only a problem in this specific case.
It would seem that the stringing issue that a lot of people have is down to wipe and retraction settings getting directly copied from other slicers that don't enforce the extruder speed as strictly, or rather prioritise finishing retraction within the wipe.
I can see how the Orca approach is probably better, but it still leaves some confused users that will think that it's a slicer problem. I certainly see a lot of them on QIDI printers users groups.
Personally I printed a retraction tower from 0 to 6mm retraction length and saw no improvement in stringing, that can lead to some false conclusions. Perhaps after slicing a warning should be displayed that the retraction cannot be completed within given wipe length?

closed 2024-06-27 by Redhorde

igiannakas, 2024-06-28:
We could add a warning - however the reality is that in most cases and in most models there will always be an area where retraction can’t complete within the given length. 

That is because the wipe length the user gives is the upper limit and in some cases where the previous line path was smaller than this (think small external perimeters, or small internal solid or sparse infill) it won’t be enough to retract fully during the wipe, especially if you’re printing fast.

take a look here: https://github.com/SoftFever/OrcaSlicer/pull/3094

It explains what the issue can be if for whatever reason your model commands a very short wipe distance. 

In addition some users have a very long wipe distance set up which in my view is detrimental to quality - I have mine set to 1mm. Any more than this and it can cause scarring on external surfaces I’ve found as you’re moving a dry nozzle over a printed area.

From my experience, stringing issues are best addressed through using **spiral z hop** - enable arc moves in klipper and set your z hop to spiral lift. It is very efficient at breaking the strings before moving to a new location. What you’re seeing is happening because z hop is set to traditional :)

igiannakas, 2024-06-28:
PS. You could point them to this thread if you come across this point in the Qidi groups to explain what is going on and why and also what they can do about it (spiral Z) if you can :)

frc29, 2024-08-31:
> PS. You could point them to this thread if you come across this point in the Qidi groups to explain what is going on and why and also what they can do about it (spiral Z) if you can :)

Hi, 

i have tried almost all options, but with new 2.1.1 version it seems like an impossible mission to not produce stringing, I just can't find a reliable setting for an Ender v3 KE, do you have any suggestion? The default settings also string like crazy.

When I switch to cura, same material, same temp and same retraction model, no stringing at all. 

it's really annoying because I really want to continue using Cura

igiannakas, 2024-08-31:
Set retract before wipe to 70%. Set wipe distance to 1mm. Disable reduce infill retractions. Enable wipe on loops and wipe before external perimeter. Enable ramp z hop.

frc29, 2024-09-01:
> Set retract before wipe to 70%. Set wipe distance to 1mm. Disable reduce infill retractions. Enable wipe on loops and wipe before external perimeter. Enable ramp z hop.

[image]

I activated the options you mentioned, but still same result, attaching picture of the settings, it's same retraction speed and length than cura 5.8.1

these are cura settings
[image]

igiannakas, 2024-09-01:
Enable retract on layer change. Also what is the lift z enforcement section content? Is the printer actually z hopping?

finally role based wipe speed should be on.

igiannakas, 2024-09-01:
Also what test are you running to validate the results?

frc29, 2024-09-01:
> Also what test are you running to validate the results?

[image]

this is the test i'm using, I also used the normal one with only 2 "columns"
https://www.thingiverse.com/thing:3420677 

PS: thanks for the assistance! I'm going to try these settings

frc29, 2024-09-01:
> Also Make sure your travel speed is fast! Moving slowly when z hopping can create strings.

my printer is a Ender 3 V3 KE, for Orca I have 400 mm/s and on Cura 300 mm/s

[image]

this is the difference, right -> orca, left -> cura

frc29, 2024-09-06:
> my printer is a Ender 3 V3 KE, for Orca I have 400 mm/s and on Cura 300 mm/s [...]

2 days later of testing different settings, drying up filament, still the same. It's a shame because there are a lot of people affected by this on reddit.
left -> orca
right -> Cura (with almost no setting at all, just using the same settings crom creality slicer)
[image]

Guess I'll just have to go back to Cura, shame really.

beaverdude, 2024-09-23:
So is there any solution for Orca yet? I played with every possible speed-retraction-wipe setting and can't get rid of a spider web like on the left pic above