OrcaSlicer/OrcaSlicer #5904: Measure tool not available when part is below build plate

bigwoof, 2024-06-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

14.5 (23F79)

### Printer

Generic - Ender3 clone

### How to reproduce

1. add stl part onto build plate
2. move part below build plate (intention is to only print form a certain height upwards - it is a quick and dirty way to modify models when you are too lazy to modify at source)
3. press "U" to invoke measure tool - it and it does not work (greyed out)  

### Actual results

Before moving part:
[image]
After moving part:
[image]

### Expected results

I think that the slicer should still be able to measure parts when they are clipped by the build plate. 

### Project file & Debug log uploads

[needle holder.3mf.zip](https://github.com/user-attachments/files/16021982/needle.holder.3mf.zip)
[logs.zip](https://github.com/user-attachments/files/16022000/logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Loving the slicer and continuous development with the latest features!

sleewok, 2024-07-19:
I can confirm that this is an issue. I am also encountering this.

fritzw, 2024-10-15:
The most annoying part is that this occurs if *any* part on *any* build plate is below the surface, and it does not even tell you why the button is disabled. I was very confused about this yesterday when working on a large project file, and only found the cause just now after I accidentally moved an object down and noticed the Measure button disabling itself.

fritzw, 2025-01-14:
Not stale.

sleewok, 2025-04-15:
Definitely not

fritzw, 2025-07-18:
@sleewok I just tested it in 2.3.0 on Linux and could not reproduce the issue. Does it still exist for you? Otherwise it should be closed.

closed 2025-10-25 by github-actions[bot]