OrcaSlicer/OrcaSlicer #5906: Prepare area completely blank when using Compiz on Linux

torriem, 2024-06-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Fedora 40, Mate Spin

### Additional system information

Using Compiz as the window manager with the Mate Desktop.  Graphics card is AMD Radion Vega.

### Printer

Irrelevant

### How to reproduce

1.  Run Compiz as the compositing window Manager under X.org session
2. Run OrcaSlicer, not for the first time.

This problem does not occur when using the Marco window manager, so the problem is a specific interaction between OrcaSlicer and Compiz.  Interestingly, if I delete the configuration files in .config, and run OrcaSlicer, it does load and shows the prepare plate just fine. But subsequent loads will show a blank prepare area.

### Actual results

Prepare area of the window is completely blank

### Expected results

Prepare area should show the build plate

### Project file & Debug log uploads

I can attach the output of stderr, but it does not shed light on this problem.  All that's printed out are various GTK warnings that come from wxWidgets.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

This is only a problem between Compiz and OrcaSlicer, as near as I can tell.  Under other window managers and desktop environments there is no issue.

and1gum, 2024-07-02:
Same issue with Win11 and AMD 24.6.1 drivers, it is possible to force "prepare" tab to the normal view - when changing the language, application does "rebuild" and displays the tab normally. But after closing and reopening - black tab again.

torriem, 2024-07-03:
Interesting.  I didn't think to try that on Linux.  I will next time I get a chance.

bepstein111, 2024-07-11:
Can confirm, Debian 12 on Ryzen 5800X, NVIDIA 3070Ti, 32GB RAM, KDE + Compiz + Emerald = no graphics on prepare tab, but I can see all the menus and settings on the left side, just no 3D. GCode preview Tab works fine.
Once I switch back over to kwin, I need to quit out and re-open OrcaSlicer, but then it works fine.
I will note that both PrusaSlicer and BambuStudio work fine under compiz.

bepstein111, 2024-08-02:
I tried playing around with application profiles in the NVIDIA control panel to no avail. The similar thread for AMD drivers seems to have solved their issue by turning off sharpening in the driver, but the same thing with NVIDIA doesn't seem to help.

adolfotregosa, 2024-08-19:
does this help ??

__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ./OrcaSlicer_Linux_Ubuntu2404_V2.2.0-dev_c6d7155092af9b224253666b778b5e75.AppImage

eteq, 2024-08-24:
@adolfotregosa - thank you so much!  The ``__EGL_VENDOR_LIBRARY_FILENAMES`` envar did it for me.

(Note I am not on compiz, but rather on hyprland, but otherwise the problem is the same)

phdmetallurgist, 2024-09-15:
> does this help ??
> 
> __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ./OrcaSlicer_Linux_Ubuntu2404_V2.2.0-dev_c6d7155092af9b224253666b778b5e75.AppImage

@adolfotregosa 
+1 upvote

This solved my graphics card issue with OrcaSlicer - Thank you!
Fedora Workstation 40
Nvidia GTX 3060 driver 560.35.03
OrcaSlicer v2.1.1

The solution that worked:
run this in the CLI:
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ./OrcaSlicer_Linux_V2.1.1.AppImage 

I suspect that it will also work with Bambu Studio as well since I was having the same issue with both.

sbman, 2024-10-21:
Adding the env variable worked with Kubuntu 24.10 and an Nvidia 3080 TI, after I had resolved an issue with a gtk library, and also downloaded the latest beta nightly build.

Booom775, 2025-03-10:
> __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ./OrcaSlicer_Linux_Ubuntu2404_V2.2.0-dev_c6d7155092af9b224253666b778b5e75.AppImage
This is still an issue with current releases for me. Running Pop-os 22.04, the environment variable trick does work but means that the home page is fully blank and windows such as adding and changing filament settings and printers are blank.