OrcaSlicer/OrcaSlicer #5907: OrcaSlicer v 2.1.1 (latest release) Windows App crashes after slicing up to 80%

esrpradeep, 2024-06-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Creality ender 3 V2 with Sprite Extruder (Direct Drive)

### How to reproduce

1. Install new version of OrcaSlicer (from 2.1.0 to 2.1.1)
2. Open existing project and click Slice
3. Application closes after reaching 80%
4. Go back to OrcaSlicer 2.1.0 
5. Repeat step 2. Open existing project and slice; it doesn't crash and slices successfuly

### Actual results

Application crashes when attempted to slice any model.
This issue is observed only after upgrading to version 2.1.1 and in portable file.

### Expected results

Should slice or expected to work as before without any crash

### Project file & Debug log uploads

[Nozzle temperature test.zip](https://github.com/user-attachments/files/16027047/Nozzle.temperature.test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lunatic1972, 2024-06-28:
I realize I'm not on windows(Manjaro linux--Latest) however my version also crashes after slicing and the software is sitting idle for a bit(2-3 minutes) and it becomes unresponsive and I'm required to close it out forcefully

arionkrause, 2024-06-28:
It's happening to me as well.

closed 2024-10-05 by github-actions[bot]