OrcaSlicer/OrcaSlicer #5917: Support Top Z Distance is incorrect with Normal (Snug) supports and Thick Bridges enabled

acarpenter628, 2024-06-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Creality K1

### How to reproduce

1. Open a model that requires supports (example: https://www.thingiverse.com/thing:2823038) (Rotate it so that it's actually useful as a support test)
2. Configure Settings:
- Quality:
  - Layer Height: 0.2
  - First Layer Height: 0.2
  - Thick bridges: Enabled
- Support:
  - Enable support
  - Support Type: Normal(auto)
  - Support Style: Snug
    - I'm not 100% sure it won't happen on Grid/Default style, but I'm able to reproduce it consistently with Snug style
  - Top Z Distance: 0.2
  - Independent support layer height: Disabled
    - I'm also able to reproduce on another model where this setting is enabled but doesn't make a difference
3. Slice 

### Actual results

0.4mm distance between support interface and bridge.  In attached project, see layers 38-41.

Note that the preview shows "thick bridges" as 2 layers tall, so visually it looks like a 0.2mm gap, but the thick bridge isn't printed until 0.4mm higher than the support interface

[image]
[image]
[image]

### Expected results

0.2mm distance between support interface and bridge

### Project file & Debug log uploads

[thick_bridge_support_bug.zip](https://github.com/user-attachments/files/16037403/thick_bridge_support_bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Seems like an off-by-one error - setting Top Z Distance to 0.4mm creates a 0.6mm distance
Note:
- This is not reproducible when using Support Type: Tree(auto) - the distance is 0.2mm as expected
- This is also not reproducible when Thick Bridges is disabled

CCS86, 2024-09-29:
I'm having this same issue. I'm asking for 0.2mm distance and getting 0.61.

@SoftFever 

https://github.com/user-attachments/assets/d407f34f-e8a0-4ab9-bc44-389a61e39fe7

CCS86, 2024-12-21:
@SoftFever 

Here is an even more clear illustration. When [thick bridges] is enabled, the support distance is being calculated by the "thickened" bottom surface. IMO, the bottom surface should be calculated as [nozzle height] - [layer height], ignoring the extra thickness when calculating the support top interface:

Normal:

[image]

Thick bridges:

[image]

CCS86, 2024-12-21:
This is very weird, I just disabled [detect overhang walls] and got this:

[image]

Then turned it back on and got this (which looks correct):

[image]

CCS86, 2025-12-27:
@SoftFever  @igiannakas any chance you guys could look into this?