OrcaSlicer/OrcaSlicer #5922: Add multiple exclusion areas for bed clips

knightslaw, 2024-06-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I upgraded to a G10 build plate that is held down with clips. My build space is small 180x180. So being able to accurately exclude 4 different areas would make love so much better!

### Which printers will be beneficial to this feature?

Marlin, All

### Describe the solution you'd like

The solution should be able to mark off small sections on the build plate where the clips are in the way. Slicer understands where these areas are and be able to route the nozzle around these areas . Only the small areas were the clips are will be avoided. The space in-between the clips can still be used as the not waste build space

### Additional context

[image]

foreachthing, 2024-06-29:
You could model your build plate (3D, stl) and use that as your "real" build plate. I've the same for our Ultimaker 2, which has these clips as well. BUT, arrange could have issues with a non-rectangular build plate.
Or, you can draw an svg and use this image on a rectangular build plate and manage these areas manually. That way arrange will work better. That's what I use, currently.

tlhintoq, 2024-07-01:
Or make a PNG the size of the print bed and put your clips marked out on it, then make that image the texture for the printer.
That way it always comes up automatically as part of the printer definition.
I did that to add the printer model here so there was just no way I could slice for the wrong machine.
[image]

oldcrazyeye, 2024-08-02:
The problem with just adding an svg,png ect. is the toolhead will still travel across the masked zones which will cause collisions still if you need mask off for structural collisions like toolchangers and awd configurations

zenduin, 2024-08-15:
This is already possible, you just need to learn how to express it as points of a polygon.
Here is my crazy example:
[image]
0x0, 12x0, 12x30, 2x30, 2x54, 0x54, 0x140, 234x140, 234x146, 230x146, 230x140, 0x140, 0x250, 20x250, 20x305, 190x305, 190x345, 275x345, 275x330, 350x330, 350x350, 0x350

Here is an example of 30x20mm exclusion zones at the corners of a 200x200mm build plate:
[image]

0x0, 30x0, 30x20, 0x20, 0x180, 30x180, 30x200, 170x200, 170x180, 200x180, 200x20, 170x20, 170x0, 200x0, 200x200, 0x200

pterodaktil02, 2024-11-07:
I make  zone as  described 

0x0, 10x0, 10x300, 290x300, 290x0, 300x0, 300x300, 0x300  
 
 
 but  when i add   model  and say fill  the bed   it say that  its to close  to exclusion area
[image]

knightslaw, 2024-11-27:
@zenduin This was great Thank you!! the only thing is I am also getting the error @pterodaktil02 is getting.
[image]

EchotreePrints, 2024-12-08:
Greetings Programs!
I found this issue after modifying my K1 to have a heatsink attached to the extruder motor, and was looking for ways to make more than one exclusion zone in Orca Slicer. @zenduin made a great post to generate multiple zones, but it seems as if the solution isn't perfect.
    
If I outline 4 squares, 20mm in length from each corner on each side, it seems like the slicer actually excludes 20mm from the entire side of the build plate.

The coordinates used: `0x0, 0x20, 20x20, 20x0, 200x0, 200x20, 220x20, 220x200, 200x200, 200x220, 20x220, 20x200, 0x200, 0x220, 220x220, 0x220` This should start a square area at 0x0, to 20x20, then follow an awkward path around each of the squares on the inside of the plate, before closing off the polygon on the outside, see image for clarification.
[image]
This creates what looks like a backwards exclusion zone, where I can only print where I don't want to, but if you look, you can see that it's not throwing an error, and will let me slice the plate, and if I move an object into the places I don't want to print, it throws the error like expected. (weird)

[image]
[image]

What frustrates me is that even with this awkward geometry, it seems to exclude the entire area around the center anyways, I can't slice an object that is less than 20mm from the edge, regardless of if it's in an exclusion zone or not.

[image]
[image]
[image]

I even tried making the exclusion have a non-zero width edge to see if that would fix it, but it doesn't change anything. The slicer still acts as if the entire edge is excluded.
[image]

Maybe I missed something, but I absolutely second the request for multiple independent exclusion zones. This would make setting them much easier.

dhenry437, 2024-12-18:
Anyone able to assist with creating the following? I wish to give the bed a "padding" of 10mm.

[image]

Everything I've done i just end up with triangles everywhere. Or is there a better way to do this in Orca?

Thanks

gilly991, 2025-01-03:
It's already enabled in the toolset!! ... you just need to create a polygon using cartesian coordinates, ie in a single direction avoiding crossing the polygon sides or origin point.  Here's my old Ender 3 V2, with std glass bed 235mm x 235mm plate, origin set at 3,3 and all around exclusion zones to suit ~8mm wide bed clips in Y and 1mm each side in X to clear the std purge line & just land on bed etc.  This gives an actual max print area of 224 x 219mm.
[image]
[image]

KopiasCsaba, 2025-03-29:
Yeah, it seems to be impossible, even if you get the drawing right, it will warn about "Cube is too close to exclusion area, there may be collisions when printing.

I have attached a 3mf file, hopefully containing everything. But, the key is the exclusion area (build plate 200x200+0+0).
0x0, 6x0, 6x90, 0x90, 0x190, 38x190, 38x200, 178x200, 178x190, 200x190, 200x90, 197x90, 197x0, 200x0, 200x200, 0x200, 0x0

And i have placed little cubes around the exclusion zones.

[Cube.3mf.txt](https://github.com/user-attachments/files/19521346/Cube.3mf.txt) (rename from .txt to .3mf)

For now i have drawn it as an svg, but it does not prevent the slicer to make a move that crosses the boundary, so it is really not a solution as I need to be careful still.

realjackwang, 2025-06-02:
> Anyone able to assist with creating the following? I wish to give the bed a "padding" of 10mm.
> 
> [image]
> 
> Everything I've done i just end up with triangles everywhere. Or is there a better way to do this in Orca?
> 
> Thanks

Here you are, 
```
0x0, 220x0, 220x220, 0x220, 0x0, 10x0, 10x210, 210x210, 210x10, 10x10, 10x0
```

It looks like the exclude bed area can only be a contiguous area and can't be set for each side individually, the padding code is generated by **ChatGPT**

closed 2026-10-22 by SoftFever