OrcaSlicer/OrcaSlicer #5932: RatRig V-Core 4 Profiles

nameisCruzCruz, 2024-06-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Hi! I'm here submitting the new V-Core 4 Profiles by RatRig! This is the first iteration of the profiles so all the feedback from the users is welcomed and will be taken into consideration for future updates!
I've attached the RatRig.Json file as well as the RatRig Folder, everything was tested and in proper working condition!

[RatRig Profiles 29July2024.zip](https://github.com/user-attachments/files/16042258/RatRig.Profiles.29July2024.zip)

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

RatRig V-Core 4 profiles! This might not be the right place to submit them. If so, sorry in advance and I will publish them in the right place.

cochcoder, 2024-07-01:
I'm not seeing much difference from the profiles in #5660, is there something I'm missing?

nameisCruzCruz, 2024-07-01:
> I'm not seeing much difference from the profiles in #5660, is there something I'm missing?

We also based the profiles in PrusaSlicer and did a few tests in-house for the different machine variants, trying to optimize as best as possible for now, while keeping the file structure organized with our developers, ensuring future updates and iterations simpler. (Hopefully!).  Unfortunately I wasn't aware of your PR, is it already on OrcaSlicer? I couldn't see it in my local installation

cochcoder, 2024-07-01:
It should be in the latest version as it was included in v2.1 RC and shows up in my installation of v2.1.1

cochcoder, 2024-07-01:
I noticed that some values like `wall_count` & `inital_layer_height` have been removed completely, from parent files like `fdm_process_ratrig_common`, was this intentional?

nameisCruzCruz, 2024-07-02:
[fdm_process_ratrig_common.json](https://github.com/user-attachments/files/16064138/fdm_process_ratrig_common.json)
It wasn't intentional, I don't recall modifying it. Sorry for the trouble, here is an updated version with those corrections.

cochcoder, 2024-09-16:
@nameisCruzCruz 

So sorry for the long sudden wait, I plan to finish this very soon. I was wondering if the processes for the RatRig V-Core 4 should be named as "0.20mm QUALITY V-Core 4 @ RatRig" rather than "0.20mm QUALITY COREXY @ RatRig"?

cochcoder, 2024-09-18:
I also noticed in fdm_process_common that the support threshold angle changed from 30 degrees to 40 degrees, is this correct? Lastly, should the sparse infill pattern be changed back to crosshatch rather than grid? As SoftFever has made that the new default.

PR #5966 (2024-10-07): General improvments to RatRig V-Core 4 & V-Core 4 HYBRID
Resolves #5932

**The following changes made are made to all V-Core 4 versions** 

- Improved cover's quality 
[image]
- Added RatRig's PunkFil ABS, PETG, and PETG CF filaments
- Added PETG-CF10 filament type & RatRig to filament vendors
- Increased max z speed to 200 from 50 and max acceleration travel to 10,000 from 9,000
- Enabled wipe and z hop, as well as, increased retract before wipe to 70% from 0%
- Improved layer change G-Code
- Fixed pause G-Code
- Added extruder clearance parameters 

*Almost all of these changes come from @na [... cut]
closed 2024-10-07 by SoftFever