OrcaSlicer/OrcaSlicer #5934: Wont open, crashes on Mac 10.15.7

cntbill, 2024-06-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer V2.1.1 Official Release

### Operating System (OS)

macOS

### OS Version

Mac OS Catalina v10.15.7

### Additional system information

Model Name:	Mac Pro
  Model Identifier:	MacPro5,1
  Processor Name:	6-Core Intel Xeon
  Processor Speed:	3.46 GHz
  Number of Processors:	2
  Total Number of Cores:	6
  L2 Cache (per Core):	256 KB
  L3 Cache:	12 MB
  Hyper-Threading Technology:	Enabled
  Memory:	32 GB
  Boot ROM Version:	144.0.0.0.0
  SMC Version (system):	1.39f11
  SMC Version (processor tray):	1.39f11
  Serial Number (processor tray):	J503407RECZJB     
  Hardware UUID:	B724234B-5D12-579C-BEE7-BEEE32CB8E85

Display Card info:

NVIDIA GeForce GT 640:

  Chipset Model:	NVIDIA GeForce GT 640
  Type:	GPU
  Bus:	PCIe
  Slot:	Slot-1
  PCIe Lane Width:	x16
  VRAM (Total):	2 GB
  Vendor:	NVIDIA (0x10de)
  Device ID:	0x0fc0
  Revision ID:	0x00a2
  ROM Revision:	MacVidCards
  Metal:	Supported, feature set macOS GPUFamily1 v4
  Displays:
Cinema HD:
  Display Type:	LCD
  Resolution:	1920 x 1200 (WUXGA - Widescreen Ultra eXtended Graphics Array)
  UI Looks like:	1920 x 1200
  Framebuffer Depth:	24-Bit Color (ARGB8888)
  Display Serial Number:	2A7203GNXMN 
  Main Display:	Yes
  Mirror:	Off
  Online:	Yes
  Rotation:	Supported
  Automatically Adjust Brightness:	No
HP 25es:
  Resolution:	1920 x 1080 (1080p FHD - Full High Definition)
  UI Looks like:	1920 x 1080 @ 60 Hz
  Framebuffer Depth:	24-Bit Color (ARGB8888)
  Display Serial Number:	3CM649032Z   
  Mirror:	Off
  Online:	Yes
  Rotation:	Supported
  Automatically Adjust Brightness:	No

### Printer

NOT applicable 

### How to reproduce

Click on OrcaSlicer and just get error... OrcaSlicer Quit Unexpctedly... so don't even open

### Actual results

Process:               OrcaSlicer [24706]
Path:                  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
Identifier:            com.softfever3d.orca-slicer
Version:               2.1.1 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OrcaSlicer [24706]
User ID:               501

Date/Time:             2024-06-29 19:22:48.056 -0400
OS Version:            Mac OS X 10.15.7 (19H2)
Report Version:        12
Anonymous UUID:        A98F4D78-B17C-1ABE-F309-5EFFA4FDDAEC

Sleep/Wake UUID:       F5FBE6C3-876E-4FE6-93B9-65B0F5EA4872

Time Awake Since Boot: 3300000 seconds
Time Since Wake:       2300 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000000d449000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [24706]

VM Regions Near 0xd449000:
--> 
    __TEXT                 000000010d449000-0000000111a95000 [ 70.3M] r-x/r-x SM=COW  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer

Application Specific Information:
/Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.softfever3d.orca-slicer   	0x000000010d50349e __cxx_global_var_init.609 + 14
1   dyld                          	0x00000001198a41d3 ImageLoaderMachO::doModInitFunctions(ImageLoader::LinkContext const&) + 535
2   dyld                          	0x00000001198a45de ImageLoaderMachO::doInitialization(ImageLoader::LinkContext const&) + 40
3   dyld                          	0x000000011989effb ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 493
4   dyld                          	0x000000011989d0b4 ImageLoader::processInitializers(ImageLoader::LinkContext const&, unsigned int, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 188
5   dyld                          	0x000000011989d154 ImageLoader::runInitializers(ImageLoader::LinkContext const&, ImageLoader::InitializerTimingList&) + 82
6   dyld                          	0x000000011988b6a8 dyld::initializeMainExecutable() + 199
7   dyld                          	0x0000000119890bba dyld::_main(macho_header const*, unsigned long, int, char const**, char const**, char const**, unsigned long*) + 6667
8   dyld                          	0x000000011988a227 dyldbootstrap::start(dyld3::MachOLoaded const*, int, char const**, dyld3::MachOLoaded const*, unsigned long*) + 453
9   dyld                          	0x000000011988a025 _dyld_start + 37

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff6b482b68 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff6b482b68 start_wqthread + 0

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000119920268  rbx: 0x000000010d503490  rcx: 0x00007ffee27b6c08  rdx: 0x00007ffee27b6ba0
  rdi: 0x0000000000000002  rsi: 0x00007ffee27b6b88  rbp: 0x00007ffee27b2640  rsp: 0x00007ffee27b2630
   r8: 0x0000000119920330   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x00007ffee27b2650  r14: 0x000000000d449000  r15: 0x0000000000000000
  rip: 0x000000010d50349e  rfl: 0x0000000000010202  cr2: 0x000000000d449000
  
Logical CPU:     2
Error Code:      0x00000004 (no mapping for user data read)
Trap Number:     14
....... and so on

### Expected results

that program would start

### Project file & Debug log uploads

dont work

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Version 2.0 worked fine

bmberetta, 2024-06-30:
I have a similar issue, but on MacOS 11.7.10. System info below

Model Name:	MacBook Pro
  Model Identifier:	MacBookPro11,2
  Processor Name:	Quad-Core Intel Core i7
  Processor Speed:	2.2 GHz
  Number of Processors:	1
  Total Number of Cores:	4
  L2 Cache (per Core):	256 KB
  L3 Cache:	6 MB
  Hyper-Threading Technology:	Enabled
  Memory:	16 GB
  System Firmware Version:	478.0.0.0.0
  SMC Version (system):	2.18f15

Process:               OrcaSlicer [14510]
Path:                  /Volumes/*/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
Identifier:            com.softfever3d.orca-slicer
Version:               2.1.1 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OrcaSlicer [14510]
User ID:               501

Date/Time:             2024-06-30 16:51:15.248 -0400
OS Version:            macOS 11.7.10 (20G1427)
Report Version:        12
Anonymous UUID:        9DAB9000-BB62-18DB-B3DA-23442A0C144E

Sleep/Wake UUID:       439673DA-B080-4EE6-89F1-EBD1875BA389

Time Awake Since Boot: 81000 seconds
Time Since Wake:       300 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000000f47c000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [14510]

VM Regions Near 0xf47c000:
--> 
    __TEXT                      10f47c000-113ac8000    [ 70.3M] r-x/r-x SM=COW  /Volumes/*/OrcaSlicer.app/Contents/MacOS/OrcaSlicer

Application Specific Information:
/Volumes/OrcaSlicer 2/OrcaSlicer.app/Contents/MacOS/OrcaSlicer

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.softfever3d.orca-slicer   	0x000000010f53649e __cxx_global_var_init.609 + 14
1   dyld                          	0x000000011e23db47 ImageLoaderMachO::doModInitFunctions(ImageLoader::LinkContext const&) + 535
2   dyld                          	0x000000011e23df52 ImageLoaderMachO::doInitialization(ImageLoader::LinkContext const&) + 40
3   dyld                          	0x000000011e238ae6 ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 492
4   dyld                          	0x000000011e23689f ImageLoader::processInitializers(ImageLoader::LinkContext const&, unsigned int, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 191
5   dyld                          	0x000000011e236940 ImageLoader::runInitializers(ImageLoader::LinkContext const&, ImageLoader::InitializerTimingList&) + 82
6   dyld                          	0x000000011e2228b1 dyld::initializeMainExecutable() + 199
7   dyld                          	0x000000011e229041 dyld::_main(macho_header const*, unsigned long, int, char const**, char const**, char const**, unsigned long*) + 9098
8   dyld                          	0x000000011e221224 dyldbootstrap::start(dyld3::MachOLoaded const*, int, char const**, dyld3::MachOLoaded const*, unsigned long*) + 450
9   dyld                          	0x000000011e221025 _dyld_start + 37

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff20775420 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff20775420 start_wqthread + 0

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x000000011e2c4298  rbx: 0x000000010f536490  rcx: 0x00007ffee0783ba8  rdx: 0x00007ffee0783b40
  rdi: 0x0000000000000001  rsi: 0x00007ffee0783b30  rbp: 0x00007ffee077e9c0  rsp: 0x00007ffee077e9b0
   r8: 0x000000011e2c4360   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x00007ffee077e9d0  r14: 0x000000000f47c000  r15: 0x0000000000000000
  rip: 0x000000010f53649e  rfl: 0x0000000000010206  cr2: 0x000000000f47c000
  
Logical CPU:     2
Error Code:      0x00000004 (no mapping for user data read)
Trap Number:     14

Thread 0 instruction stream:
  30 0f 11 4c 30 10 48 8d-04 f8 48 39 f9 74 27 4c  0..L0.H...H9.t'L
  89 f2 4d 8d 24 fc 66 66-66 2e 0f 1f 84 00 00 00  ..M.$.fff.......
  00 00 49 8b 0c 24 48 89-08 49 83 c4 08 48 83 c0  ..I..$H..I...H..
  08 49 39 d4 75 ec 49 89-41 08 48 83 c4 18 5b 41  .I9.u.I.A.H...[A
  5c 41 5d 41 5e 41 5f 5d-c3 4c 89 ef e8 61 f1 ff  \A]A^A_].L...a..
  ff 90 55 48 89 e5 41 56-53 4c 8b 35 c2 1d 59 04  ..UH..AVSL.5..Y.
 [41]80 3e 00 74 05 5b 41-5e 5d c3 48 8b 1d c0 6b  A.>.t.[A^].H...k	<==
  59 04 48 89 df e8 88 40-95 01 48 8d 3d 11 43 95  Y.H....@..H.=.C.
  01 48 8d 15 3a 5b f4 ff-48 89 de e8 c2 82 a6 03  .H..:[..H.......
  41 c6 06 01 5b 41 5e 5d-c3 66 0f 1f 84 00 00 00  A...[A^].f......
  00 00 55 48 89 e5 41 56-53 4c 8b 35 7a 1d 59 04  ..UH..AVSL.5z.Y.
  41 80 3e 00 74 05 5b 41-5e 5d c3 48 8b 1d 78 6b  A.>.t.[A^].H..xk
  
Thread 0 last branch register state not available.

Binary Images:
       0x10f47c000 -        0x113ac7fff +com.softfever3d.orca-slicer (2.1.1 - ???) <1D2DC878-C28C-3EDC-9AB8-952173B47DE7> /Volumes/*/OrcaSlicer.app/Contents/MacOS/OrcaSlicer

Noisyfox, 2024-07-03:
Because since 2.1.1 we have upgraded to Xcode 15, which no longer support macos 11 and earlier.

cntbill, 2024-07-03:
Well that Sucks.... So I will just have to stick with the older version and hope 2.1.0 will work ok or just use a slicer that will work with my Mac.... bummmer

Noisyfox, 2024-07-03:
It seems to be a bug in Xcode 15.0 (which is what we are using for our CI build), and this bug should have been fixed in Xcode 15.1.
GitHub announced that they will use Xcode 15.4 since Jul 22, which should fix this issue then (hopefully).

Noisyfox, 2024-07-03:
I've raised a PR with new Xcode version, could you please give it a go once it finishes?

cntbill, 2024-07-03:
Sure I will give it a try... About when do you think it will be finished ?

Noisyfox, 2024-07-03:
Maybe in ~20mins.

bmberetta, 2024-07-03:
I just tried the build of 4 hours ago from the nightlies. Still crashes, see below...

Process:               OrcaSlicer [16764]
Path:                  /Volumes/*/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
Identifier:            com.softfever3d.orca-slicer
Version:               2.1.1 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OrcaSlicer [16764]
User ID:               501

Date/Time:             2024-07-03 12:57:48.361 -0400
OS Version:            macOS 11.7.10 (20G1427)
Report Version:        12
Anonymous UUID:        9DAB9000-BB62-18DB-B3DA-23442A0C144E

Sleep/Wake UUID:       965F57DC-63CF-49DF-9309-AB1F51BD1DAE

Time Awake Since Boot: 100000 seconds
Time Since Wake:       93 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000000a8f0000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [16764]

VM Regions Near 0xa8f0000:
--> 
    __TEXT                      10a8f0000-10ef4c000    [ 70.4M] r-x/r-x SM=COW  /Volumes/*/OrcaSlicer.app/Contents/MacOS/OrcaSlicer

Application Specific Information:
/Volumes/OrcaSlicer 1/OrcaSlicer.app/Contents/MacOS/OrcaSlicer

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.softfever3d.orca-slicer   	0x000000010a9aaa1e __cxx_global_var_init.609 + 14
1   dyld                          	0x000000011dca0b47 ImageLoaderMachO::doModInitFunctions(ImageLoader::LinkContext const&) + 535
2   dyld                          	0x000000011dca0f52 ImageLoaderMachO::doInitialization(ImageLoader::LinkContext const&) + 40
3   dyld                          	0x000000011dc9bae6 ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 492
4   dyld                          	0x000000011dc9989f ImageLoader::processInitializers(ImageLoader::LinkContext const&, unsigned int, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 191
5   dyld                          	0x000000011dc99940 ImageLoader::runInitializers(ImageLoader::LinkContext const&, ImageLoader::InitializerTimingList&) + 82
6   dyld                          	0x000000011dc858b1 dyld::initializeMainExecutable() + 199
7   dyld                          	0x000000011dc8c041 dyld::_main(macho_header const*, unsigned long, int, char const**, char const**, char const**, unsigned long*) + 9098
8   dyld                          	0x000000011dc84224 dyldbootstrap::start(dyld3::MachOLoaded const*, int, char const**, dyld3::MachOLoaded const*, unsigned long*) + 450
9   dyld                          	0x000000011dc84025 _dyld_start + 37

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff20775420 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff20775420 start_wqthread + 0

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x000000011dd27298  rbx: 0x000000010a9aaa10  rcx: 0x00007ffee530fba8  rdx: 0x00007ffee530fb40
  rdi: 0x0000000000000001  rsi: 0x00007ffee530fb30  rbp: 0x00007ffee530a9c0  rsp: 0x00007ffee530a9b0
   r8: 0x000000011dd27360   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x00007ffee530a9d0  r14: 0x000000000a8f0000  r15: 0x0000000000000000
  rip: 0x000000010a9aaa1e  rfl: 0x0000000000010206  cr2: 0x000000000a8f0000
  
Logical CPU:     4
Error Code:      0x00000004 (no mapping for user data read)
Trap Number:     14

cntbill, 2024-07-03:
Same Here...

Process:               OrcaSlicer [43521]
Path:                  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
Identifier:            com.softfever3d.orca-slicer
Version:               2.1.1 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OrcaSlicer [43521]
User ID:               501

Date/Time:             2024-07-03 13:04:04.957 -0400
OS Version:            Mac OS X 10.15.7 (19H2)
Report Version:        12
Anonymous UUID:        A98F4D78-B17C-1ABE-F309-5EFFA4FDDAEC

Sleep/Wake UUID:       9B57E094-1E87-4A24-8C74-ECE2342CB053

Time Awake Since Boot: 3500000 seconds
Time Since Wake:       450 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000007ae8000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [43521]

VM Regions Near 0x7ae8000:
--> 
    __TEXT                 0000000107ae8000-000000010c134000 [ 70.3M] r-x/r-x SM=COW  /Applications/OrcaSlicer 2.app/Contents/MacOS/OrcaSlicer

Application Specific Information:
/Applications/OrcaSlicer 2.app/Contents/MacOS/OrcaSlicer

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.softfever3d.orca-slicer   	0x0000000107ba249e __cxx_global_var_init.609 + 14
1   dyld                          	0x00000001151171d3 ImageLoaderMachO::doModInitFunctions(ImageLoader::LinkContext const&) + 535
2   dyld                          	0x00000001151175de ImageLoaderMachO::doInitialization(ImageLoader::LinkContext const&) + 40
3   dyld                          	0x0000000115111ffb ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 493
4   dyld                          	0x00000001151100b4 ImageLoader::processInitializers(ImageLoader::LinkContext const&, unsigned int, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 188
5   dyld                          	0x0000000115110154 ImageLoader::runInitializers(ImageLoader::LinkContext const&, ImageLoader::InitializerTimingList&) + 82
6   dyld                          	0x00000001150fe6a8 dyld::initializeMainExecutable() + 199
7   dyld                          	0x0000000115103bba dyld::_main(macho_header const*, unsigned long, int, char const**, char const**, char const**, unsigned long*) + 6667
8   dyld                          	0x00000001150fd227 dyldbootstrap::start(dyld3::MachOLoaded const*, int, char const**, dyld3::MachOLoaded const*, unsigned long*) + 453
9   dyld                          	0x00000001150fd025 _dyld_start + 37

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff6b482b68 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff6b482b68 start_wqthread + 0

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000115193268  rbx: 0x0000000107ba2490  rcx: 0x00007ffee8117c00  rdx: 0x00007ffee8117b98
  rdi: 0x0000000000000002  rsi: 0x00007ffee8117b80  rbp: 0x00007ffee8113640  rsp: 0x00007ffee8113630
   r8: 0x0000000115193330   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x00007ffee8113650  r14: 0x0000000007ae8000  r15: 0x0000000000000000
  rip: 0x0000000107ba249e  rfl: 0x0000000000010202  cr2: 0x0000000007ae8000
  
Logical CPU:     6
Error Code:      0x00000004 (no mapping for user data read)
Trap Number:     14

Noisyfox, 2024-07-04:
I see... So apparently it's a different bug then.

pickanothername, 2024-07-04:
I'm also experiencing this issue on a 2023 M3 Pro MacBook Pro running macOS 14.5.

Noisyfox, 2024-07-04:
> I'm also experiencing this issue on a 2023 M3 Pro MacBook Pro running macOS 14.5. Update: @Noisyfox The nightly build fixes the issue for me!

Which build more specificly?

Noisyfox, 2024-07-04:
BTW I've pushed another commit to the PR, could you give it a try? @bmberetta @cntbill

pickanothername, 2024-07-04:
> > I'm also experiencing this issue on a 2023 M3 Pro MacBook Pro running macOS 14.5. Update: @Noisyfox The nightly build fixes the issue for me!
> 
> Which build more specificly?

OrcaSlicer_Mac_arm64_V2.1.1 from yesterday.

bmberetta, 2024-07-04:
> BTW I've pushed another commit to the PR, could you give it a try? @bmberetta @cntbill

Still no luck. Tried the OrcaSlicer_Mac_arm64_V2.1.1 from the nightlies. It is marked as "yesterday" so not sure if this build contains your latest PR.

Process:               OrcaSlicer [18058]
Path:                  /Volumes/*/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
Identifier:            com.softfever3d.orca-slicer
Version:               2.1.1 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           OrcaSlicer [18058]
User ID:               501

Date/Time:             2024-07-04 15:05:39.125 -0400
OS Version:            macOS 11.7.10 (20G1427)
Report Version:        12
Anonymous UUID:        9DAB9000-BB62-18DB-B3DA-23442A0C144E

Sleep/Wake UUID:       AFC6552E-74E5-4349-9480-D99DDA761324

Time Awake Since Boot: 120000 seconds
Time Since Wake:       130 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000032b1000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [18058]

VM Regions Near 0x32b1000:
--> 
    __TEXT                      1032b1000-10790d000    [ 70.4M] r-x/r-x SM=COW  /Volumes/*/OrcaSlicer.app/Contents/MacOS/OrcaSlicer

Application Specific Information:
/Volumes/OrcaSlicer 2/OrcaSlicer.app/Contents/MacOS/OrcaSlicer

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.softfever3d.orca-slicer   	0x000000010336ba1e __cxx_global_var_init.609 + 14
1   dyld                          	0x0000000113669b47 ImageLoaderMachO::doModInitFunctions(ImageLoader::LinkContext const&) + 535
2   dyld                          	0x0000000113669f52 ImageLoaderMachO::doInitialization(ImageLoader::LinkContext const&) + 40
3   dyld                          	0x0000000113664ae6 ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 492
4   dyld                          	0x000000011366289f ImageLoader::processInitializers(ImageLoader::LinkContext const&, unsigned int, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 191
5   dyld                          	0x0000000113662940 ImageLoader::runInitializers(ImageLoader::LinkContext const&, ImageLoader::InitializerTimingList&) + 82
6   dyld                          	0x000000011364e8b1 dyld::initializeMainExecutable() + 199
7   dyld                          	0x0000000113655041 dyld::_main(macho_header const*, unsigned long, int, char const**, char const**, char const**, unsigned long*) + 9098
8   dyld                          	0x000000011364d224 dyldbootstrap::start(dyld3::MachOLoaded const*, int, char const**, dyld3::MachOLoaded const*, unsigned long*) + 450
9   dyld                          	0x000000011364d025 _dyld_start + 37

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff20775420 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff20775420 start_wqthread + 0

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x00000001136f0298  rbx: 0x000000010336ba10  rcx: 0x00007ffeec94eba8  rdx: 0x00007ffeec94eb40
  rdi: 0x0000000000000001  rsi: 0x00007ffeec94eb30  rbp: 0x00007ffeec9499c0  rsp: 0x00007ffeec9499b0
   r8: 0x00000001136f0360   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x00007ffeec9499d0  r14: 0x00000000032b1000  r15: 0x0000000000000000
  rip: 0x000000010336ba1e  rfl: 0x0000000000010206  cr2: 0x00000000032b1000
  
Logical CPU:     6
Error Code:      0x00000004 (no mapping for user data read)
Trap Number:     14

Noisyfox, 2024-07-05:
https://github.com/SoftFever/OrcaSlicer/pull/6002/checks
Try the artifacts from this link

cntbill, 2024-07-05:
> https://github.com/SoftFever/OrcaSlicer/actions/runs/9775636085 Try the artifacts from this link first, the changes haven't been included into nightly build yet.
> 
> And if the previous link still crashes, please try this one again and see if it works: https://github.com/SoftFever/OrcaSlicer/actions/runs/9786889513
> 
> NOTE: Please make sure to try the first link FIRST! Thanks!

Not sure what happened to my last comment... but yes the First Link, "OrcaSlicer_Mac_x86_64_PR-6002" worked for me, loaded stl, slice, and printed through IP host printer

cntbill, 2024-07-05:
Thanks! ever so much for you work!!

Noisyfox, 2024-07-05:
Great! So all we need to do is wait until Jul 22.

bmberetta, 2024-07-05:
I can confirm that "OrcaSlicer_Mac_x86_64_PR-6002" works for me as well! Thanks very much for the quick response to, and resolution of the issue!

Spillekant, 2024-09-18:
Same problem and MACOS as the original poster. 2.0.0 suddenly couldn't open a window or a new project. Up till now I've been very happy with Orca Slicer. It's alle a bit too complicated for me, a huge "Thank You" to the folks doing the legwork. I had hoped downloading a fresh 2.1.1 would help, but no. Tried PR-6002 - no. Zapping the app and reinstalling 2.0.0 got me working again. Maybe it was another issue, but I'm too tired now to try PR-6002 again

Noisyfox, 2024-09-19:
> Same problem and MACOS as the original poster. 2.0.0 suddenly couldn't open a window or a new project. Up till now I've been very happy with Orca Slicer. It's alle a bit too complicated for me, a [...]

Have you tried the 2.2.0 beta?

Spillekant, 2024-09-19:
Hi Noisyfox,
thank you for your comment. No, I have not. I’m rather pressed right now, maybe when I have more time again.

Cheers
Kurt

Spillekant, 2024-09-27:
@Noisyfox, running 2.2.0 now, seems to work, Thanks again.

closed 2025-01-03 by github-actions[bot]

KonkeBonk, 2025-04-04:
Hi, a little late to the party but I have had the same issue for the last couple of months, and finally I'll try to get help here. The latest build 2.3.0. still crashes for me after I login to my Bambu account. Is there a fix for this?