OrcaSlicer/OrcaSlicer #5937: SEMM: More than 4 filament flush_length params

53Aries, 2024-06-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently only 4 filament flush lengths are supported. Carryover from Bambu AMS.
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Param Flush_length_X should be supported up to maximum. Bambu lists up to 16 flush lengths but more is better. 
https://wiki.bambulab.com/en/software/bambu-studio/placeholder-list
Also some of the flushing function should be changed to be more generic vs being configured for a Bambu AMS system.
[image]

53Aries, 2024-06-30:
> Are you trying to use this variable for a printer running happy hare or a Bambu printer?

Neither. Tradrack mmu with custom purging. I found i can use flush_length and it will get the full length for that flush. Flush_length_x doesn't actually refer to the lane number like bambu placeholder says

igiannakas, 2024-06-30:
I’d strongly recommend you try happy hare, especially as it has an integration with blobifier already available which uses the orca purge volume.

closed 2024-10-07 by github-actions[bot]