OrcaSlicer/OrcaSlicer #5945: Unrecognizable line ranges in color scheme

Rickthebig, 2024-06-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The color scheme classifies lines layerwise according to the choosen criteria. That is a fine idea but at times quite useless as it is implemented, because in the preview of the print the individual gradation levels cannot always be recognized. For example, the line width color scheme shows line widths varying from 0.06 to 2.00 mm. I can recognize line widths of about 0.49 mm (green) but I just can't find any lines with a width of 2.00 mm. I would need to, in order to tackle unprecise extrusion. Even if I scroll the height of the previewed model up and down, such line widths remain hidden- it could be just a tiny spot (or a bug?).

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Every gradation level line in the color scheme should have a check box right to it. All check boxes would be checked by default, but in order to recognize specific gradation levels (for example, specific widths), those individual levels could be checked individually. If I want to recognize layers with lines of 2.00 width only, I would check just that level in the color  scheme, and in the preview those layers would  appear highlighted, so I know where those features are and correlated then with print defects. Otherwise it is difficult to diagnose and fix those defects, if the corresponding layers cannot be previewed.

### Describe alternatives you've considered

Add a percentage of every level right to the check box, as additional feature, for all types of criteria (speed, flow, etc.) in the color scheme.

[image]

CCS86, 2024-07-08:
For me, the bug of having the color scale max automatically set to 3 times larger than any width in the sliced print makes it essentially useless.

@SoftFever some improvements were made to this in a Superslicer release: https://github.com/supermerill/SuperSlicer/releases/tag/2.5.59.8

Having a realistic automatic maximum would be great, but manual control of the scale bounds would be even better.

SoftFever, 2024-10-07:
It makes sense.
I see if I can add it in 2.3

CCS86, 2025-01-06:
@SoftFever , did this get added to the roadmap?

dzid26, 2025-03-08:
Adjusting color range is one way. 

Another would be to ignore hidden objects.
For example I have [this](https://youtu.be/PqcEz09jQ7c?t=279) custom g-code thick purge line (it works great btw):

[image]

Even if I hide custom lines with the checkbox and it's no longer rendered, it is still taken for the colormap range calculation:
[image]

(This might be why OP can't see the thick red color. )

CCS86, 2025-06-13:
@SoftFever any progress on this?

Here is a before and after view, with the only change being that I deleted all the start gcode (and with it the purge lines). At the very least, if any hidden lines can be ignored from the scale bounds calculation, that would be a huge improvement. 

The line width display is SO much more useful when scaled properly:

[image]

[image]

yw4z, 2025-11-06:
can you check color palette in here https://github.com/SoftFever/OrcaSlicer/pull/8436
i think it also has improvements for normal vision