OrcaSlicer/OrcaSlicer #5949: Brim not printed until after first layer -causes print failure

lewis2e, 2024-06-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Not a performance issue

### Printer

Bambu X1C

### How to reproduce

Open and slice the attached file. In the preview, note how the brim is not printed until layer 3. This causes the print to fail.
OPTIONAL: Open the same file in Bambu Studio and slice without changing any settings. Note that the brim is printed on layer 1 as expected.

### Actual results

Print fails as brim is being printed in mid-air.

### Expected results

Brim should be printed on layer 1 to provide adhesion to the build plate.

### Project file & Debug log uploads

[Cerritos2.zip](https://github.com/user-attachments/files/16044057/Cerritos2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Chewieez, 2024-07-27:
I think I just experienced this issue or something very similar today. 

I downloaded a model from MakerWorld and it kept failing in the print right when it started printing the brim. In my situation only the support printed on layer 1, then the brim, printed on layer 2 and always failed to adhere to the build plate. 

I opened the same model in Bambu Studio and the brim and support both start printing on layer 1. 

This is the model I used.
https://makerworld.com/en/models/538761#profileId-455921

E-Saggese, 2024-10-05:
Seeing the issue and I think I understand the cause. My model has only an infinitesimal contact area with the plate, as it sits on two edges. This causes Orca to ignore the first layer for the object's floor and first printable layer and this also causes the brim to not start until the first printable layer in the object (still prints supports at first layer). Code should be fixed by either slicing any infinitesimal touch point as one line-wide so the brim is also printed together with it, or to print the brim even if the area in question has a zero-width part. Alternatively we could fix the "lay on face" feature so if the face has a zero-area intersection with the part to displace the part down until it has a printable surface along that plane. If I was able to code decently I would fix it myself :-).

closed 2025-01-11 by github-actions[bot]