OrcaSlicer/OrcaSlicer #5950: Disable "Auto-record Monitoring" by default and store files on local computer

Alkerion, 2024-06-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The "Auto-record Monitoring" is turned ON by default.

This create a HUGE memory usage on the printer SD Card, something really unwanted.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The "Auto-record Monitoring" must be turned OFF by default.

The files must be stored on local computer, not on printer SD card.

Add an option to choose the file location.

Alkerion, 2024-06-30:
32Gb on ipcam folder after 18h of print :
[image]

Noisyfox, 2024-07-02:
You can turn that off via bambu handy or here
[image]

Alkerion, 2024-07-02:
I know thanks, it's not the purpose of this ticket to learn how to turn it OFF.

Noisyfox, 2024-07-02:
Then raise a ticket to BambuLab, not here.

closed 2024-10-09 by github-actions[bot]