OrcaSlicer/OrcaSlicer #5954: Build failed @ Gentoo -- YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY

dsiggi, 2024-06-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Gentoo Linux

### Printer

Ender 3 with Klipper

### How to reproduce

Build OrcaSlicer:
Build line:
```
cmake -C /var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DSLIC3R_GTK=3 -DSLIC3R_PCH=0 -DSLIC3R_WX_STABLE=1 -DSLIC3R_STATIC=0 -DORCA_TOOLS=ON -DOPENVDB_FIND_MODULE_PATH=/usr/lib64/cmake/OpenVDB -DBBL_RELEASE_TO_PUBLIC=1 -DBBL_INTERNAL_TESTING=0 -DCMAKE_BUILD_TYPE=Release -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1

ninja -v -j9 -l17
```

### Actual results

Build failed with this message.

```
[98/538] /usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_LOCALE_DYN_LINK -DBOOST_LOCALE_NO_LIB -DBOOST_LOG_DYN_LINK -DBOOST_LOG_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DLIBNEST2D_GEOMETRIES_libslic3r -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DLIBNEST2D_THREADING_tbb -DOPENVDB_ABI_VERSION_NUMBER=11 -DOPENVDB_USE_DELAYED_LOADING -DSLIC3R_DESKTOP_INTEGRATION -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1_build/src/platform -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/libslic3r -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1_build/src/libslic3r -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/libnest2d/include -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/miniz -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/glu-libtess/include -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/clipper2/Clipper2Lib/include -isystem /usr/include/eigen3 -isystem /var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/libigl -isystem /usr/include/opencascade -isystem /usr/include/Imath  -O2 -pipe -fomit-frame-pointer -fext-numeric-literals -Wall -Wno-reorder -std=gnu++17 -fPIC -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -gz=zstd -MD -MT src/libslic3r/CMakeFiles/libslic3r.dir/Emboss.cpp.o -MF src/libslic3r/CMakeFiles/libslic3r.dir/Emboss.cpp.o.d -o src/libslic3r/CMakeFiles/libslic3r.dir/Emboss.cpp.o -c /var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/libslic3r/Emboss.cpp
FAILED: src/libslic3r/CMakeFiles/libslic3r.dir/Emboss.cpp.o 
/usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_LOCALE_DYN_LINK -DBOOST_LOCALE_NO_LIB -DBOOST_LOG_DYN_LINK -DBOOST_LOG_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DLIBNEST2D_GEOMETRIES_libslic3r -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DLIBNEST2D_THREADING_tbb -DOPENVDB_ABI_VERSION_NUMBER=11 -DOPENVDB_USE_DELAYED_LOADING -DSLIC3R_DESKTOP_INTEGRATION -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1_build/src/platform -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/libslic3r -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1_build/src/libslic3r -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/libnest2d/include -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/miniz -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/glu-libtess/include -I/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/clipper2/Clipper2Lib/include -isystem /usr/include/eigen3 -isystem /var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/libigl -isystem /usr/include/opencascade -isystem /usr/include/Imath  -O2 -pipe -fomit-frame-pointer -fext-numeric-literals -Wall -Wno-reorder -std=gnu++17 -fPIC -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -gz=zstd -MD -MT src/libslic3r/CMakeFiles/libslic3r.dir/Emboss.cpp.o -MF src/libslic3r/CMakeFiles/libslic3r.dir/Emboss.cpp.o.d -o src/libslic3r/CMakeFiles/libslic3r.dir/Emboss.cpp.o -c /var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/libslic3r/Emboss.cpp
In file included from /usr/include/eigen3/Eigen/Core:164,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/admesh/stl.h:31,
                 from /var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/libslic3r/Emboss.hpp:8,
                 from /var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/libslic3r/Emboss.cpp:2:
/usr/include/eigen3/Eigen/src/Core/CwiseBinaryOp.h: In instantiation of ‘Eigen::CwiseBinaryOp<BinaryOp, Lhs, Rhs>::CwiseBinaryOp(const Lhs&, const Rhs&, const BinaryOp&) [with BinaryOp = Eigen::numext::equal_to<long int>; LhsType = const Eigen::Matrix<long int, 2, 1, 2>; RhsType = const Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<double, int>, const Eigen::Matrix<double, 2, 1, 2> >; Lhs = Eigen::Matrix<long int, 2, 1, 2>; Rhs = Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<double, int>, const Eigen::Matrix<double, 2, 1, 2> >]’:
/usr/include/eigen3/Eigen/src/plugins/MatrixCwiseBinaryOps.h:45:10:   required from ‘const Eigen::CwiseBinaryOp<Eigen::numext::equal_to<typename Eigen::internal::traits<T>::Scalar>, const Derived, const OtherDerived> Eigen::MatrixBase<Derived>::cwiseEqual(const Eigen::MatrixBase<OtherDerived>&) const [with OtherDerived = Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<double, int>, const Eigen::Matrix<double, 2, 1, 2> >; Derived = Eigen::Matrix<long int, 2, 1, 2>; typename Eigen::internal::traits<T>::Scalar = long int]’
/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:291:24:   required from ‘bool Eigen::MatrixBase<Derived>::operator==(const Eigen::MatrixBase<OtherDerived>&) const [with OtherDerived = Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<double, int>, const Eigen::Matrix<double, 2, 1, 2> >; Derived = Eigen::Matrix<long int, 2, 1, 2>]’
/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1/src/libslic3r/Emboss.cpp:337:17:   required from here
/usr/include/eigen3/Eigen/src/Core/CwiseBinaryOp.h:113:7: error: static assertion failed: YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY
  113 |       EIGEN_CHECK_BINARY_COMPATIBILIY(BinaryOp,typename Lhs::Scalar,typename Rhs::Scalar);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/eigen3/Eigen/src/Core/CwiseBinaryOp.h:113:7: note: ‘Eigen::internal::has_ReturnType<Eigen::ScalarBinaryOpTraits<long int, int, Eigen::numext::equal_to<long int> > >::value’ evaluates to false
ninja: build stopped: subcommand failed.
 * ERROR: media-gfx/OrcaSlicer-2.1.1::dsiggis-gentoo-overlay failed (compile phase):
 *   ninja -v -j1 -l0 failed
 * 
 * Call stack:
 *     ebuild.sh, line  136:  Called src_compile
 *   environment, line 2537:  Called cmake_src_compile
 *   environment, line  974:  Called cmake_build
 *   environment, line  941:  Called eninja
 *   environment, line 1438:  Called die
 * The specific snippet of code:
 *       "$@" || die -n "${*} failed"
 * 
[... cut ...]
 * The ebuild environment file is located at '/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/temp/environment'.
 * Working directory: '/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1_build'
 * S: '/var/tmp/portage/media-gfx/OrcaSlicer-2.1.1/work/OrcaSlicer-2.1.1'

```

### Expected results

A working build :D.

### Project file & Debug log uploads

build.log: [build.log](https://github.com/user-attachments/files/16045690/build.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2024-07-03:
Don't build Debug config, which has lots of broken assertions. Use relwithdebinfo instead.

dsiggi, 2024-07-08:
HI,
to use the BuildLinux.sh script I have to modify it because gentoo is not an supported distro.

With the modified script I got this error.
```
┌──(siggi🦆nitro)-[/tmp/OrcaSlicer-2.1.1]
└─$ ./BuildLinux.sh -rfsc
FOUND_GTK3=/usr/lib64/libgtk-3.so.0.2409.32
Changing date in version...
done
Configuring OrcaSlicer...
cmake -S . -B build -G Ninja -DCMAKE_PREFIX_PATH=/tmp/OrcaSlicer-2.1.1/deps/build/destdir/usr/local -DSLIC3R_STATIC=1  -DBBL_RELEASE_TO_PUBLIC=1 -DBBL_INTERNAL_TESTING=0
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- No build type selected, default to Release
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_DESKTOP_INTEGRATION: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: 0
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: 0
-- CMAKE_PREFIX_PATH: /tmp/OrcaSlicer-2.1.1/deps/build/destdir/usr/local (from cache or command line)
CMake Warning at CMakeLists.txt:153 (message):
  CMAKE_PREFIX_PATH element doesn't exist:
  /tmp/OrcaSlicer-2.1.1/deps/build/destdir/usr/local

-- PROJECT_SOURCE_DIR: /tmp/OrcaSlicer-2.1.1
-- CMAKE_MODULE_PATH: /tmp/OrcaSlicer-2.1.1/cmake/modules/
-- Found PkgConfig: /usr/bin/pkg-config (found version "2.2.0") 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found DBus: /usr/include/dbus-1.0;/usr/lib64/dbus-1.0/include  
-- LIBDIR: /tmp/OrcaSlicer-2.1.1/src
-- LIBDIR_BIN: /tmp/OrcaSlicer-2.1.1/build/src
-- Found Boost: /usr/lib64/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable version "1.84.0", minimum required is "1.66") found components: system filesystem thread log locale regex chrono atomic date_time iostreams program_options 
-- TBB found in /usr/lib64/cmake/TBB
-- Found OpenSSL: /usr/lib64/libcrypto.so (found version "3.0.13")  
-- Found CURL: /usr/lib64/libcurl.so (found version "8.7.1") 
-- Found ZLIB: /usr/lib64/libz.so (found version "1.3.1")  
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.3") 
-- Found EXPAT: /usr/lib64/libexpat.so (found version "2.5.0") 
-- Found PNG: /usr/lib64/libpng.so (found version "1.6.43") 
-- Found OpenGL: /usr/lib64/libOpenGL.so   
GLEW_ROOT: /tmp/OrcaSlicer-2.1.1/deps/build/destdir/usr/local
-- Found GLEW: /usr/include (found version "2.2.0") 
-- Found NLopt in '/usr/lib64'.
-- Using NLopt include directory '/usr/include'.
-- Using NLopt library '/usr/lib64/libnlopt.so'.
COMPONENT = openvdb
-- Found Boost: /usr/lib64/cmake/Boost-1.84.0/BoostConfig.cmake (found version "1.84.0") found components: iostreams 
[... 139 lines cut ...]
                             Datum:     02:25:10 25.03.2024
                             USE:       X gstreamer libnotify opengl sdl tiff -debug -doc -pch -test -webkit ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="64 -32 -x32"
                             DEPEND     >=app-eselect/eselect-wxwidgets-20131230 dev-libs/expat[abi_x86_64(-)] media-libs/libsdl2[abi_x86_64(-)] >=dev-libs/glib-2.22:2[abi_x86_64(-)] media-libs/libjpeg-turbo:0/0.2=[abi_x86_64(-)] media-libs/libpng:0/16=[abi_x86_64(-)] sys-libs/zlib[abi_x86_64(-)] x11-libs/cairo[abi_x86_64(-)] x11-libs/gtk+:3[abi_x86_64(-)] x11-libs/gdk-pixbuf[abi_x86_64(-)] x11-libs/libSM[abi_x86_64(-)] x11-libs/libX11[abi_x86_64(-)] x11-libs/libXxf86vm[abi_x86_64(-)] x11-libs/pango[abi_x86_64(-)] media-libs/gstreamer:1.0[abi_x86_64(-)] media-libs/gst-plugins-base:1.0[abi_x86_64(-)] x11-libs/libnotify[abi_x86_64(-)] virtual/opengl[abi_x86_64(-)] media-libs/tiff:0/6=[abi_x86_64(-)] virtual/glu[abi_x86_64(-)] x11-base/xorg-proto
                             RDEPEND:   >=app-eselect/eselect-wxwidgets-20131230 dev-libs/expat[abi_x86_64(-)] media-libs/libsdl2[abi_x86_64(-)] >=dev-libs/glib-2.22:2[abi_x86_64(-)] media-libs/libjpeg-turbo:0/0.2=[abi_x86_64(-)] media-libs/libpng:0/16=[abi_x86_64(-)] sys-libs/zlib[abi_x86_64(-)] x11-libs/cairo[abi_x86_64(-)] x11-libs/gtk+:3[abi_x86_64(-)] x11-libs/gdk-pixbuf[abi_x86_64(-)] x11-libs/libSM[abi_x86_64(-)] x11-libs/libX11[abi_x86_64(-)] x11-libs/libXxf86vm[abi_x86_64(-)] x11-libs/pango[abi_x86_64(-)] media-libs/gstreamer:1.0[abi_x86_64(-)] media-libs/gst-plugins-base:1.0[abi_x86_64(-)] x11-libs/libnotify[abi_x86_64(-)] virtual/opengl[abi_x86_64(-)] media-libs/tiff:0/6=[abi_x86_64(-)] >=sys-libs/glibc-2.38-r10
                             BDEPEND:   >=app-eselect/eselect-wxwidgets-20131230 virtual/pkgconfig
                             EAPI:      7
                             Version:   3.2.2.1-r4(3.2-gtk3)^t{gpkg}
                             Datum:     18:41:02 30.06.2024
                             USE:       X curl gstreamer libnotify lzma opengl sdl spell tiff wayland webkit -debug -doc -keyring -pch -test ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="64 -32 -x32"
[... cut ...]
     Beste Versionen/Slot:   3.0.4-r5(3.0) 3.0.5.1-r1(3.0-gtk3)^t 3.2.2.1-r4(3.2-gtk3)^t{gpkg}
     Startseite:             https://wxwidgets.org/
     Finden offener Bugs:    https://bugs.gentoo.org/buglist.cgi?quicksearch=x11-libs%2FwxGTK
     Beschreibung:           GTK version of wxWidgets, a cross-platform C++ GUI toolkit
     Lizenz:                 wxWinLL-3 GPL-2 doc? ( wxWinFDL-3 )

┌──(siggi🦆nitro)-[/tmp/OrcaSlicer-2.1.1]
└─$ eselect wxwidgets list
Available wxWidgets profiles:
  [1]   gtk2-unicode-3.0
  [2]   gtk3-unicode-3.0-gtk3
  [3]   gtk3-unicode-3.2-gtk3 *

Noisyfox, 2024-07-09:
You have to use wxWidgets from the deps. Make sure you build deps first by passing '-d' option to the script.

closed 2024-10-15 by github-actions[bot]