OrcaSlicer/OrcaSlicer #5955: Application becomes unusable if at anypoint you attempt to minimize it and then restore the window

MostHated, 2024-06-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Ubuntu 23.10

### Additional system information

<details>
<summary>System Details</summary>

OS: Ubuntu 23.10 x86_64
Host: X670E Taichi
Kernel: 6.8.4-060804-generic
Uptime: 52 mins
Packages: 4587 (dpkg), 471 (nix-user), 48 (nix-default), 79 (flatpak), 26 (snap)
Shell: zsh 5.9
Resolution: 1920x1080
DE: GNOME 45.2
WM: Mutter
WM Theme: Yaru-viridian-dark
Theme: Yaru-viridian-dark [GTK2/3]
Icons: Yaru-blue [GTK2/3]
Terminal: WarpTerminal
CPU: AMD Ryzen 9 7950X (32) @ 5.881GHz
GPU: AMD ATI Radeon RX 6800/6800 XT / 6900 XT
GPU: AMD ATI 5a:00.0 Raphael
Memory: 14120MiB / 63405MiB

</details>

### Printer

Bambu Lab x1c

### How to reproduce

At any time, on any window, hit the Minimize button in the titlebar. Then attempt to restore the window via clicking on the icon in the application/task bar.

### Actual results

The 3d viewport no longer works.
Attempting to change main tabs sort of works, but the contents of the tab's view sometimes show, and if it does, part of it shows, or it doesn't respond to input.
In general, the application overall becomes unusable and must be closed and started again.

### Expected results

The application should restore fully functional.

### Project file & Debug log uploads

[orca_logs.zip](https://github.com/user-attachments/files/16045988/orca_logs.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Below is a video clip showing this occurring. It occurs no matter what, any project (or blank new window with no project), all prior versions of the application I have tried.

[minimize_example.webm](https://github.com/SoftFever/OrcaSlicer/assets/3442753/05a3fd59-0d6d-470a-8b99-0b7d7e91458a)

L2on1, 2024-07-01:
Could be the same that : #997 .

MostHated, 2024-07-01:
It's possible it's related, but doesn't seem like it would be. I never have any sort of slowness at all, until I minimize and restore the window. Though, at that point, everything gets messed up, not just the print settings tabs.

MostHated, 2024-09-30:
Just want to make sure this doesn't fall off. I still deal with this issue constantly.

closed 2025-01-07 by github-actions[bot]