OrcaSlicer/OrcaSlicer #5969: Slicing yields "No such file or directory" when another user did slicing earlier on the same date

aleksandarsusnjar, 2024-07-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04 LTS

### Additional system information

We have multiple family members use OrcaSlicer with our 3D printer. When one of us slices a model on a particular date, OrcaSlicer creates a "date" directory inside `/tmp/orcasclider_mode/`, then some directory that appears like time of day inside of it, and then more inside that.

If another person/user then tries to use OrcaSlicer using his/her own account, they get greeted with a failure pop-up that corresponds to the following error message:

```
[error] 2024-07-01 12:13:52.298803[Thread 0x000075ba9a0006c0]:Failed to create missing sub_path/tmp/orcaslicer_model/Mon_Jul_01/11_46_28#23614#49/Metadata: boost::filesystem::create_directories: Permission denied [system:13]: "/tmp/orcaslicer_model/Mon_Jul_01/11_46_28#23614#49/Metadata", "/tmp/orcaslicer_model/Mon_Jul_01/11_46_28#23614#49"
```

I personally can `sudo` my way around the problem with `chown` and/or `chmod` ... but I also have to pre-create some other expected folders manually - OrcaSlicer won't do it, probably because of the location of the check relative to the occurrence of the error.

I've tried to:

1. Find existing posts with these errors.
2. Find existing posts mentioning `orcaslicer_model`
3. Find existing posts mentioning `tmp` and OrcaSlicer
4. Looked at raw OrcaSlicer configuration files for ways to configure separate directories for each user.
5. Looked for ways to configure dedicated, user-specific `tmp` folders for OrcaSlicer

... and failed to find any solutions.

I wanted to upgrade to a newer version but it is failing on my Ubuntu 24.04 LTS and I can't get it to work due to the library issues.

### Printer

Bambu Labs X1C

### How to reproduce

1. Set up a (Linux?) multiuser system with at least two users and OrcaSlicer available for them.
2. Log in as one of those users and slice something.
3. **Switch** user to the other one. You likely may need to leave the OrcaSlicer running but I didn't check.
4. Open OrcaSlicer and try slicing something.

### Actual results

Too late for a screenshot TODAY - we're slicing something, can't mess with that now to resproduce it. I may update this tomorrow. However, I did supply the log:

```
```
[error] 2024-07-01 12:13:52.298803[Thread 0x000075ba9a0006c0]:Failed to create missing sub_path/tmp/orcaslicer_model/Mon_Jul_01/11_46_28#23614#49/Metadata: boost::filesystem::create_directories: Permission denied [system:13]: "/tmp/orcaslicer_model/Mon_Jul_01/11_46_28#23614#49/Metadata", "/tmp/orcaslicer_model/Mon_Jul_01/11_46_28#23614#49"
```
```

### Expected results

No errors, successful slicing, that does not conflict with what other system users are doing.

### Project file & Debug log uploads

[OrcaSlicer-bug-report.zip](https://github.com/user-attachments/files/16057097/OrcaSlicer-bug-report.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

aleksandarsusnjar, 2024-07-01:
Quick code search yields the following match in `src/libslic3r/Model.cpp`:

```cpp
// BBS: backup dir
std::string Model::get_backup_path()
{
    if (backup_path.empty())
    {
        auto pid = get_current_pid();
        boost::filesystem::path parent_path(temporary_dir());
        std::time_t t = std::time(0);
        std::tm* now_time = std::localtime(&t);
        std::stringstream buf;
        buf << "/orcaslicer_model/";
        buf << std::put_time(now_time, "%a_%b_%d/%H_%M_%S#");
        buf << pid << "#";
        buf << this->id().id;

        backup_path = parent_path.string() + buf.str();
```

That `temporary_dir()` appears to come via `utils.cpp` from the following `OrcaSlicer.cpp`:

```cpp
int CLI::run(int argc, char **argv)
{
    ...
    ...
    BOOST_LOG_TRIVIAL(info) << "finished setup params, argc="<< argc << std::endl;
    std::string temp_path = wxFileName::GetTempDir().utf8_str().data();
    set_temporary_dir(temp_path);
```

I am not familar with how to control `wxFileName::GetTempDir()` to get a user-specific temp directory path.

aleksandarsusnjar, 2024-07-01:
Screenshot available... This one is coming from 2.1.1.
[image]

aleksandarsusnjar, 2024-07-02:
My present workaround: create a script to start OrcaSlicer (which I placed in `/usr/local/bin/orcaslicer` for my convenience:

```sh
#!/bin/sh

(
  BASE=~/OrcaSlicer
  TEMP=$BASE/tmp
  WORK=$BASE/work

  mkdir --parents $TEMP
  mkdir --parents $WORK
  
  cd $WORK && TMP=$TEMP TMPDIR=$TEMP TEMPDIR=$TEMP /usr/local/bin/OrcaSlicer_Linux_V2.1.1.AppImage
)
```
... use that instead. You may want to change the `/usr/local/bin/` to wherever your AppImage is.

While at it, my `Orca Slicer.desktop` file is as follows:

```
[Desktop Entry]
Name=Orca Slicer
Comment=The app that slices sliceable things that can be sliced for slicing.
Exec=/usr/local/bin/orcaslicer
Type=Application
Version=2.1.1
Terminal=false
Icon=/usr/local/bin/orca-slicer_220764.webp
Categories=Application;
```

I downloaded the icon from where I could find it.

closed 2024-10-09 by github-actions[bot]

jrha, 2025-12-19:
`pam-tmpdir` provides another workaround for this issue by automatically managing per-user temporary directories.