OrcaSlicer/OrcaSlicer #5973: Thumbnails not being displayed on qidi x-max 3 screen

pmbroth, 2024-07-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

windows

### Printer

Qidi X-max 3

### How to reproduce

send job to printer, no thumbnail generated
[image]

### Actual results

black screen, no thumbnail is shown, goto Qidi users forums, users are all stating this with 2.1.1

### Expected results

Thumbnail to be displayed

### Project file & Debug log uploads

any stl will produce this outcome

settings below

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]

kiniad, 2024-07-01:
I could confirm this for the Qidi X-Plus 3.
Reason: The COLPIC section (gimage, simage) have to be at the end of the gcode, as it was before 2.1 final and the thumpnails revamp.
See: https://github.com/SoftFever/OrcaSlicer/pull/5555

Viktor-N-snr, 2024-07-04:
QIDI X-Plus 3, yes this problem is still there since previous versions. I tried all sorts of combinations of resolution, number of resolutions (for example 380x380/COLPIC only), other bitmap formats and so on. Nothing helped at all.
A small thumbnail will appear in the file list. But not in the main window. Just the black square instead of the preview. With QIDI slicer it works corectly of course, ORCA doesn't.
QIDI slicer settings are: 380x380/QIDI, 210x210/QIDI, 110x110/PNG.
Well, ORCA doesn't know the QIDI type and uses COLPIC instead, but of course I immediately tried 380x380/COLPIC, 210x210/COLPIC, 110x110/PNG but alas — still the same, just black squere and not white circle or — required thumbnail… :-)))

cableguyed, 2024-07-14:
Verified that if you save the gcode file, open with notepadd++ or similar and cut the ;gimage and ;simage sections, paste them at the very end, the thumbnails come right up when you use the device screen to upload the gcode file. I haven't had a chance to actually print a test file yet, but it looks promising. Should be a simple fix for Orca code.

Viktor-N-snr, 2024-07-14:
Thanks for this interesting solution, but really there is no chance to edit each G-Code, when I am sending that directly to the printer. :-)

cableguyed, 2024-07-15:
Hi,
If you click on the down arrow next to the print tab and select save as
gcode instead, then create the file, edit it to make it show the thumbnail,
then go to the device tab and under jobs click on the plus sign and select
upload, then browse to the edited gcode file and you will have the
thumbnail on the printer screen when you select to print the file.
This is just a temporary fix until the coders get around to qidi update on
orca... I have multiple qidi printers and like to see what is on each at a
glance.
Regards,
Ed

cableguyed, 2024-07-15:
> This is what I did, and it works on my qidi x-max 3, I moved the thumbnail
> section to the bottom of the configuration file.
> fdm_qidi_x3_common.json

That made no difference for my X-Max3.

nicolalamacchia, 2024-07-25:
Use [this](https://gist.github.com/nicolalamacchia/1f60a4b8e8ca4ac499b1d877ae373c12) script in the post-processing setting: Global Settings -> Others -> [scroll to the bottom] -> Post-processing Scripts -> type the full path of the script you downloaded (make sure that the script is executable).

It worked for my X-Plus 3.

Thanks @kiniad. Looks like moving the whole thumbnail block works.

jimwhitelaw, 2024-08-14:
I tried the solution posted by @pmbroth; it did not work on my X-Max 3.

cableguyed, 2024-08-18:
Compiling the python script (with minor tweak to get the start point fixed) works great for both my X-Max3 and X-Smart 3.
Thanks Nicola!

ovidakis, 2024-08-25:
> Compiling the python script (with minor tweak to get the start point fixed) works great for both my X-Max3 and X-Smart 3. Thanks Nicola!

Hi @cableguyed ! How did u do that? I'v copied that file to my: /home/mks/qidi_thumbnail.py and make it executable and after that I copied address to orcaslicer. Still not workig. Am I missing something? 

[image]
[image]
Thank you!

cableguyed, 2024-08-25:
First, make a copy of the qidi_thumbnail.py just in case...
I had to edit the qidi_thumbnail.py to change the    " # Find the start and
end of the thumbnail block" section to:
    for i, line in enumerate(lines):
        if re.match(r'; HEADER_BLOCK_END', line):
            thumbnail_start = i+1
        if re.match(r'; external perimeters*', line):
            thumbnail_end = i-1
I haven't done any Python programming at all and any serious programming
for years so I had to google some python methods to figure out that the
program wants to look in the Python/scripts area for the compiled result.
Below is my file path once I created the executable in Python. (And put
your path in the post-processing scripts area in Orca.)

C:\Users\cable\AppData\Local\Programs\Python\Python312\Scripts\dist\qidi_thumbnail.exe;

I also had to change the image sizes for the different LCD screen sizes
between my X-Max3 and my X-Smart3.
X-Max3 = 380x380/COLPIC, 210x210/COLPIC, 110x110/PNG
X-Smart3 = 210x210/COLPIC, 110x110/COLPIC, 110x110/PNG

Hope this helps!
Regards,
Ed

ovidakis, 2024-08-26:
> First, make a copy of the qidi_thumbnail.py just in case... I had to edit the qidi_thumbnail.py to change the " # Find the start and end of the thumbnail block" section to: for i, line in [...]

Hi Ed, still not woking :( I looked to g-code file and I found at the end of the file that the post-processing script done the job. Probably xindi/display cant read correctly those lines. By the way, my printer (X-max3) is on latest firmware based on Debian 12 Bookworm.

cableguyed, 2024-08-26:
I only update firmware from qidi direct, don't know if that makes a
difference.

Brandon520x, 2024-09-11:
I can confirm the thumbnails are not working on my X-Plus 3 with current QIDI firmware using 2.1.1.  Very frustrating since this seems to have been committed in the latest build.

astrojolo, 2024-09-17:
Still not working on the X-Plus 3 and 2.2.0-beta version.

toxin9993, 2024-10-02:
still not  working on the X-Plus3 
Orcaslicer v2.2.0-beta2

pmbroth, 2024-10-09:
rc 22 still thumbnails not working. .

breathless19, 2024-10-18:
I'm in the same boat as everyone else.... thumbnails still not working on the latest OrcaSlicer 2.2 RC

pmbroth, 2024-10-30:
@breathless19 You edited the script that you downloaded from his site?  Then you just placed the whole script in the post processing area?

pmbroth, 2024-10-30:
I firgured it out. I compoiled the script and all works...

pmbroth, 2024-10-31:
Thank you again for your solution!!!! @cableguyed

cptcl, 2024-12-06:
And we cannot have a simple solution for this?

BobW55, 2025-01-03:
Sorry I am late to the party.
If I read everything correctly, I am using a X Max 3 with Orca 2.2.0
I have downloaded the Python script and edited it using Thoney.  (Still a novice python person)
What are you all using to convert the script to an EXE file?  And I assume I need to post a link to the file here?
[image]

cableguyed, 2025-01-03:
Hi Bob,
It has been a while since I did it but you need to download an IDE in order
to compile the script into an exe. There are several, search for python
IDE's. I had to tweak the script a little in order to get it to capture the
start and end of the 2 image sections. Look at a gcode file from your
slicer for your printer and find unique lines at the start and end of the
images and edit the script to work properly. Again, it has been some time
since I did it so I'm a little hazy on the specifics but if I could do it,
I'm sure you can too. And yes, put the exe file path in the location you
referenced. It still works even with several firmware updates since I did
it. Just make sure to put the path in if you change to a new setup for your
printer.

BobW55, 2025-01-03:
I managed to get the PY into an EXE file.   I can see it run and I get no returned error.
Is this the proper location for the .json file?
[image]

BobW55, 2025-01-04:
Ok found the correct place in the Orca slicer folder to place the Json file.   Have to wait for current print to finish so I can reboot everything.   Fingers crossed.

BobW55, 2025-01-04:
Double checked that I put the .json file and the correct spot and that it has been edited to move the thumbnails to the last few lines of code in the file.
Must be something corrupt in the EXE file I made.

cableguyed, 2025-01-04:
I didn't have to do anything with a json file, just put the .exe that was
compiled from the script. I did have to tweak the script to trigger on the
start and end lines of the image sections of the gcode file. Then put the
file path of that .exe in the place in the slicer for post processing
scripts.

BobW55, 2025-01-04:
Yes I added the i+1  and the i-1 as you indicated then I compiled it.

cableguyed, 2025-01-04:
Then all you may need to do is leave the .exe file wherever the complier
saved it and copy the file path and paste in the slicer location.

BobW55, 2025-01-04:
I appreciate the help.   Recompiled it.
now getting this error:
C:\Users\k8pos\AppData\Local\Temp\.OrcaSlicer.upload.cba5-46b0-0e15-e854 failed.
Error code: -1
I'll just wait till orca fixes it.

cptcl, 2025-01-10:
not possible to just upload the compiled working exe somewhere so we can download and use it?

cableguyed, 2025-01-10:
I had to tweak the original script to grab the start and end lines of the
image sections so I doubt it would work. I tested by first saving out a
gcode file instead of printing it, then made a copy of that and went in
with notepad++ and copied the 2 adjoining image sections from the top area
of the file and pasted it at the end of the file. I used a small quick
print and tried the unmodded one, no image on the preview screen, then the
modded one and saw the preview. I then looked at the script that was posted
and saw that the triggers would not work for my gcode setup and changed it
to grab the first line before the top image section and then the last line
of the second image section. I had never used python before so I downloaded
it and it didn't have an easy (to me) way to compile into an executable so
I searched for and easy IDE for it and that allowed me to compile. I just
left the exe where python put it and just copied the path and pasted it
into the postprocessing script section of ORCA.
I'm old and my memory isn't all that great but that was the process I used.
It is still working for both my X-Max3 and X-Smart3. I hope this helps. In
the meantime I will try to gather the IDE and tweaks I did, I just don't
know if I kept the original script to see the changes.
Regards,
Ed

cableguyed, 2025-01-10:
I used pycharm to compile (I think). Here is the original script and the
script I had to mod to make it work on my gcode files. You may have to
tweak it for your setup.

BobW55, 2025-01-10:
Probably need the .json file and the exe file for it to work.

cableguyed, 2025-01-10:
exe
https://drive.google.com/file/d/19w7T0AAwF6UPFbI3i3VuNAxwpZKZTBF3/view?usp=drive_link
don't remember any json needed but i do have one banging around.

BobW55, 2025-01-11:
So I downloaded the **fdm_qidi_x3_common.json** file from github.  The placement of the thumbnail code is at the end of the file.
[https://github.com/SoftFever/OrcaSlicer/blob/main/resources/profiles/Qidi/machine/fdm_qidi_x3_common.json](url)
On my machine it was located towards the middle.   The file resides here on your PC.
C:\Program Files\OrcaSlicer\resources\profiles\Qidi\machine\fdm_qidi_x3_common.json
I also DL the exe file that cableguyed was so kind to share.  I placed mine in the root of where all my QIDI projects are.
I now have working thumbnails.

Thank you Cableguyed ever so much.  
I hope the developers add this fix to the next release.

cableguyed, 2025-01-11:
Glad to hear it is working! When you change profiles or something, just
look at the postprocessing box to make sure the path is still there. I keep
a copy of the path up on my notepad++ so I can always pop it in when needed.
Regards,
Ed

closed 2025-04-20 by github-actions[bot]