OrcaSlicer/OrcaSlicer #5978: "Can't add prime tower if variable layer height is not the same"... but variable layer height is not active

Arakon, 2024-07-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Win 11

### Printer

Bambu A1

### How to reproduce

- Load the project file
- Note the error message

### Actual results

[image]

### Expected results

It should slice as no variable layer height is selected. Manually resetting layer height also seems to make no difference. This might be related to this issue in original BS code: https://github.com/bambulab/BambuStudio/issues/3641

### Project file & Debug log uploads

[A1toolBOX (1).zip](https://github.com/user-attachments/files/16061718/A1toolBOX.1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DBendit, 2024-07-04:
I'm running into the same issue (on the same project file, funnily enough). Notably, it loads fine in Bambu Studio v1.9.3.50, which backs up the theory that it's related to the linked upstream issue.

53Aries, 2024-07-05:
I have this issue if I'm using a layer height modifier on a print with multiple parts of different colors.

adminmat, 2024-07-22:
Having the same issue today. Tried aligning the layer heights still erroring. Orca 2.1.1

closed 2024-10-28 by github-actions[bot]

pm0u, 2025-11-29:
This is still happening to me on 2.3.1, it appears someone noted it happening upstream on 2.3.0 - not sure if this is related to that or something else in orca.

If all objects on the plate are the same it seems to work. If there is a different object somewhere in there, it fails - even if the modified height and settings are the same across all objects on the plate.

Art-Porto, 2026-10-26:
A fix for this is proposed in #16462. The prime tower check now compares the layers the objects are actually sliced at, instead of their layer height profiles, so it no longer refuses a plate where variable layer height is not in use, or where only some objects are different, as long as they are sliced on the same layers.

If you'd like to try it, test builds for Windows, macOS and Linux are attached to the PR's checks: [How to download PR artifacts](https://www.orcaslicer.com/wiki/how_to_download_pr_artifacts). A comment on the PR saying whether it works with your project would help the review.