OrcaSlicer/OrcaSlicer #5980: SBS filament type

AbrieViljoen, 2024-07-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No SBS filament type to select from the "Type" dropdown under Basic information section of Filament settings.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Under Filament settings menu, on the Filament tab in the Basic Settings area where it says "Type", could an option please be added for "SBS" filament type ?

AbrieViljoen, 2024-07-11:
If possible also add "PLA+"

dgauche, 2024-07-29:
@AbrieViljoen I have added SBS for Bambu printers and will soon create a pull request for that.

Please let me know what printers you have.

dgauche, 2024-07-29:
@AbrieViljoen see pull request [here](https://github.com/SoftFever/OrcaSlicer/pull/6278).

AbrieViljoen, 2024-07-29:
> @AbrieViljoen I have added SBS for Bambu printers and will soon create a pull request for that.
> 
> Please let me know what printers you have.

Thanks for helping with this request. I have the Creality K1C and a Creality V3 KE. Can you also add the PLA+ profile for these two printers?

dgauche, 2024-07-29:
Hi @AbrieViljoen this pull request should allow you to create your own custom filaments based on SBS.

[image]

I will add SBS to more printers as soon as I have some time.

I am however not sure that it makes sense to have PLA+ as a separate filament.  It is after all just PLA with some additives. If you start adding PLA+ then perhaps PLA-HF (High flow) should also be added?

Does your settings for PLA vs PLA+ change significantly?

From a user experience I can see that it makes sense to be able to select the exact filament type that you have, but it may have the opposite effect by overwhelming users with too many filaments and there is no material difference between them.

closed 2024-11-05 by github-actions[bot]