OrcaSlicer/OrcaSlicer #5987: Orca produces a brim around model even if it's not nesessary

abychko, 2024-07-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

macOS

### OS Version

Sonoma 14.5 (23F79)

### Printer

QIDI X-Plus3

### How to reproduce

1. Added a box to print (192*228*35)
2. Selected 0.28mm Extra Draft profile (Brim is Auto) 
3. Selected PETG profile (shrinkage is 100%)

### Actual results

The box is sliced with large brim around. PETG is not Nylon, why this brim is auto-added?

[image]

### Expected results

the box is large enough to stick to the printbed, I didn't expect so large brim around large flat-bottomed model

### Project file & Debug log uploads

[Box (1).3mf.zip](https://github.com/user-attachments/files/16067201/Box.1.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Duffy77, 2024-08-27:
Same here. I habe a big object which needed no brim. Sliced with petg and auto brim.
A big brim is placed around.
Bambu studio doesn't do this with the same setting

xokahag, 2024-10-19:
I suspect the small protrusions to support the hinges trigger the auto brim activation. I think this is one of the cases where the slicer simply cant know if a brim is actually required and errs on the side of caution.

maddhin, 2025-01-06:
I am just having the same problem but Orca did not place a brim around an object and all of sudden is. I have been calibrating a filament - maybe some settings got mixed up? Anyway to re-set (default) settings so this is not happening anymore? It's really annoying as I just wanted to switch from Bambu Studio to Orca and now can't because of this weird issue:(

closed 2025-04-14 by github-actions[bot]