OrcaSlicer/OrcaSlicer #5989: Custom bed types for Bambu printers

t3chguy, 2024-07-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Given more bed types than bambu officially sells have started hitting the market, e.g. https://darkmoon3d.com/products/darkmoon-ice-build-plate-for-bambu-lab-x1-p1-a1-printers it'd be nice for a user to be able to create a matching type locally, and control some overrides based on that type being selected.

This could also allow the user to disable bed plate detection on a per-bed-type setting rather than on a per-print setting for compatibility with 3rd party plates lacking the ArUco tag.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

It should allow me to create new bed types which mimic the built in ones for ArUco & z-offset settings but allow me to specify filament temperature overrides per material type, either as an offset (e.g. -20C for PLA) or absolute values (e.g. 35C for PLA)

### Describe alternatives you've considered

1. Duplicating all of my filament profiles with one for this special textured plate with lower bed temps

### Additional context

Related to https://github.com/SoftFever/OrcaSlicer/issues/5054

Slippinjimmie, 2024-07-02:
A life changer is implemented

t3chguy, 2024-07-02:
@Slippinjimmie did you join Github to upvote & comment on my issue :D?

Slippinjimmie, 2024-07-03:
> @Slippinjimmie did you join Github to upvote & comment on my issue :D?

Yes i want this to gain traction so I jumped in lol.

closed 2024-10-09 by github-actions[bot]