OrcaSlicer/OrcaSlicer #5991: Cannot build 2.1.1 from source

oddlama, 2024-07-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

NixOS 24.11 unstable

### Printer

N/A

### How to reproduce

Try to build from source with 2.1.1 release and gcc 13.3.0.

### Actual results

Observed error log is attached below. PR #4831 introduced the issue, reverting it together with a dependent change #5843 fixes the issue.

```
[ 92%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/Utils/FixModelByWin10.cpp.o
In file included from /build/source/src/slic3r/GUI/Monitor.hpp:51:
/build/source/src/slic3r/GUI/SelectMachine.hpp: At global scope:
/build/source/src/slic3r/GUI/SelectMachine.hpp:67: warning: 'int Slic3r::GUI::get_brightness_value(wxImage)' defined but not used [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wunused-function-Wunused-function8;;]
   67 | static int get_brightness_value(wxImage image) {
      | 
In file included from /build/source/src/slic3r/GUI/AmsMappingPopup.hpp:26:
/build/source/src/slic3r/GUI/GUI_Utils.hpp:43: warning: 'Slic3r::ColorRGBA Slic3r::GUI::decode_color_to_float_array(std::string)' defined but not used [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wunused-function-Wunused-function8;;]
   43 | static ColorRGBA decode_color_to_float_array(const std::string color)
      | 
In file included from /nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/bits/shared_ptr_base.h:59,
                 from /nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/bits/shared_ptr.h:53,
                 from /nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/condition_variable:45,
                 from /build/source/src/slic3r/pchheader.hpp:25,
                 from /build/source/build/src/slic3r/CMakeFiles/libslic3r_gui.dir/cmake_pch.hxx:5,
                 from <command-line>:
/nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/bits/unique_ptr.h: In instantiation of 'std::__detail::__unique_ptr_t<_Tp> std::make_unique(_Args&& ...) [with _Tp = Slic3r::form_file; _Args = {Slic3r::form_file}; __detail::__unique_ptr_t<_Tp> = __detail::__unique_ptr_t<Slic3r::form_file>]':
/build/source/src/slic3r/Utils/Http.cpp:384:0:   required from here
/nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/bits/unique_ptr.h:1070:30: error: use of deleted function 'Slic3r::form_file::form_file(Slic3r::form_file&&)'
 1070 |     { return unique_ptr<_Tp>(new _Tp(std::forward<_Args>(__args)...)); }
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/build/source/src/slic3r/Utils/Http.cpp:89: note: 'Slic3r::form_file::form_file(Slic3r::form_file&&)' is implicitly deleted because the default definition would be ill-formed:
   89 | struct form_file
      | 
/build/source/src/slic3r/Utils/Http.cpp:89: error: use of deleted function 'boost::filesystem::basic_ifstream<charT, traits>::basic_ifstream(const boost::filesystem::basic_ifstream<charT, traits>&) [with charT = char; traits = std::char_traits<char>]'
In file included from include/boost/config.hpp:61,
                 from include/boost/algorithm/clamp.hpp:26,
                 from /build/source/src/slic3r/pchheader.hpp:59:
include/boost/filesystem/fstream.hpp:98:5: note: declared here
In file included from /build/source/src/slic3r/GUI/AmsMappingPopup.hpp:26:
/build/source/src/slic3r/GUI/GUI_Utils.hpp: At global scope:
/build/source/src/slic3r/GUI/GUI_Utils.hpp:43: warning: 'Slic3r::ColorRGBA Slic3r::GUI::decode_color_to_float_array(std::string)' defined but not used [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wunused-function-Wunused-function8;;]
   43 | static ColorRGBA decode_color_to_float_array(const std::string color)
      | 
In file included from /build/source/src/libslic3r/GCode.hpp:26,
                 from /build/source/src/libslic3r/calib.hpp:5,
                 from /build/source/src/libslic3r/AppConfig.hpp:12,
                 from /build/source/src/libslic3r/PresetBundle.hpp:5,
                 from /build/source/src/slic3r/GUI/GUI_App.hpp:17,
                 from /build/source/src/slic3r/GUI/MultiTaskManagerPage.hpp:4,
                 from /build/source/src/slic3r/GUI/MultiTaskManagerPage.cpp:1:
/build/source/src/libslic3r/GCode/SmallAreaInfillFlowCompensator.hpp:12: warning: 'Slic3r::SmallAreaInfillFlowCompensator' has a field 'std::unique_ptr<{anonymous}::tk::spline> Slic3r::SmallAreaInfillFlowCompensator::flowModel' whose type has internal linkage [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wsubobject-linkage-Wsubobject-linkage8;;]
   12 | class SmallAreaInfillFlowCompensator
      | 
[ 92%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/Utils/EmbossStyleManager.cpp.o
[ 93%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/Utils/FontConfigHelp.cpp.o
In file included from /nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/x86_64-unknown-linux-gnu/bits/c++allocator.h:33,
                 from /nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/bits/allocator.h:46,
                 from /nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/bits/alloc_traits.h:39,
                 from /nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/condition_variable:44:
/nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/bits/new_allocator.h: In instantiation of 'void std::__new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = Slic3r::form_file; _Args = {Slic3r::form_file}; _Tp = Slic3r::form_file]':
/nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/bits/alloc_traits.h:538:17:   required from 'static void std::allocator_traits<std::allocator<_Tp1> >::construct(allocator_type&, _Up*, _Args&& ...) [with _Up = Slic3r::form_file; _Args = {Slic3r::form_file}; _Tp = Slic3r::form_file; allocator_type = std::allocator<Slic3r::form_file>]'
/nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/bits/deque.tcc:170:30:   required from 'std::deque<_Tp, _Alloc>::reference std::deque<_Tp, _Alloc>::emplace_back(_Args&& ...) [with _Args = {Slic3r::form_file}; _Tp = Slic3r::form_file; _Alloc = std::allocator<Slic3r::form_file>; reference = Slic3r::form_file&]'
/build/source/src/slic3r/Utils/Http.cpp:317:0:   required from here
/nix/store/zc0nsv23pakbafngjy32kvhfzb16as43-gcc-13.3.0/include/c++/13.3.0/bits/new_allocator.h:191:11: error: use of deleted function 'Slic3r::form_file::form_file(Slic3r::form_file&&)'
  191 |         { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /build/source/src/libslic3r/GCode.hpp:26,
                 from /build/source/src/libslic3r/calib.hpp:5,
                 from /build/source/src/slic3r/GUI/DeviceManager.hpp:17,
                 from /build/source/src/slic3r/GUI/MultiMachine.hpp:5,
                 from /build/source/src/slic3r/GUI/SendMultiMachinePage.hpp:5,
                 from /build/source/src/slic3r/GUI/SendMultiMachinePage.cpp:1:
/build/source/src/libslic3r/GCode/SmallAreaInfillFlowCompensator.hpp:12: warning: 'Slic3r::SmallAreaInfillFlowCompensator' has a field 'std::unique_ptr<{anonymous}::tk::spline> Slic3r::SmallAreaInfillFlowCompensator::flowModel' whose type has internal linkage [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wsubobject-linkage-Wsubobject-linkage8;;]
   12 | class SmallAreaInfillFlowCompensator
      | 
```

### Expected results

Compilation should succeed

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2024-07-02:
Could you try building this branch and see if that works? https://github.com/Noisyfox/OrcaSlicer/tree/bugfox/gcc-13.3

oddlama, 2024-07-02:
Thanks, I can confirm that this fixes the issue :)

PR #5992 (2024-07-07): Try fixing build with gcc 13.3 (SoftFever/OrcaSlicer#5991)
Attempt to fix #5991
closed 2024-07-07 by SoftFever