OrcaSlicer/OrcaSlicer #5999: Tree support crosses into model

lin-ycv, 2024-07-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11 22621.3810

### Printer

FLsun V400

### How to reproduce

1. Open project file
2. change printer to flsun v400
3. slice

### Actual results

[image]

### Expected results

supports to be generated without crossing into model

### Project file & Debug log uploads

[SubCompactSimpleGenericv3.zip](https://github.com/user-attachments/files/16076932/SubCompactSimpleGenericv3.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

foreachthing, 2024-07-03:
Your project does not slice with tree supports...
But, to solve your issue; You have to "assemble" all the parts that are interfering with supports of other parts.
Easiest solution: Assemble all the parts. Done.

lin-ycv, 2024-07-15:
@foreachthing while that does work, assembling parts together mean i can't use exclude object feature. I do think the slicer should be able to know if supports are coliding with prints without assembling?

Here's the proper project with tree support
[SubCompactSimpleGenericv3_withTrees.zip](https://github.com/user-attachments/files/16233673/SubCompactSimpleGenericv3_withTrees.zip)

closed 2024-10-21 by github-actions[bot]