OrcaSlicer/OrcaSlicer #6005: Dropdowns not working on Arch Wayland

NicolaiVdS, 2024-07-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Arch Linux x86_64

### Additional system information

OS: Arch Linux x86_64 
Kernel: 6.9.7-arch1-1 
DE: Hyprland
CPU: Intel i9-10900 (20) @ 5.200GHz 
GPU: NVIDIA GeForce RTX 3070 
Memory: 7840MiB / 15909MiB 

### Printer

Creality Ender 3

### How to reproduce

/

### Actual results

when clicking a dropdown it wont open

### Expected results

that the drop down opens

### Project file & Debug log uploads

[debug_Wed_Jul_03_18_01_39_945733.log.0.zip](https://github.com/user-attachments/files/16087628/debug_Wed_Jul_03_18_01_39_945733.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Nvidia driver version 555.58

MMOOBB, 2024-07-04:
Getting the same issue on my machine with an AMD GPU

MMOOBB, 2024-07-04:
https://github.com/hyprwm/Hyprland/issues/6698 could be related to this if you're using wayland

MMOOBB, 2024-07-04:
Can confirm, downgrading hyprland to 0.41.1 fixes this issue. It should be fixed in the next update https://github.com/hyprwm/Hyprland/issues/6616

Skaifer, 2024-07-04:
Same behaviour on Bazzite 40; KDE 6.1.1. Funny thing, the dropdowns were working today, but the app kept randomly crashing while I was switching trough tabs. And homepage is all grey, moving cursor over makes xwayland crash again (this causes Vivaldi to close as well). So I figured I should check .desktop file, does it include WEBKIT_DISABLE_COMPOSITING_MODE=1. Now it appears, that I don't have to use this variable to have app working properly. But something changed in this process, causing dropdowns to stop working. I'll have to change all the settings manually until this is solved :(

Maxb0tbeep, 2024-07-04:
same behavior for me, on hyprland

Skaifer, 2024-07-05:
Yesterday I did all the slicing on X11 session. Today, dropdowns are working completely fine on Wayland. Go figure what caused it.

hirn-lego, 2024-07-08:
I ran into the same issue with Qtile 0.26.0 on wayland
hyprland 0.41.1 works fine

Distro is Manjaro

lividhen, 2024-07-14:
> Can confirm, downgrading hyprland to 0.41.1 fixes this issue. Hopefully fixed in the next update [hyprwm/Hyprland#6616](https://github.com/hyprwm/Hyprland/issues/6616)

How did you go about downgrading? I built the release source from the github but the color channels seem to be swapped for me because everything is orange...

felipebalbi, 2024-07-14:
Same issue here, Arch Linux hyprland 0.41.2.

virtualspan, 2024-07-23:
This issue was on Hyprland's side, at the time of making, it has been fixed in the latest git release, but not stable release yet.

Sources:
- [Issue marked as resolved on Hyprland's Github](https://github.com/hyprwm/Hyprland/issues/6698#issue-2380081202)
- [The commit solving the issue](https://github.com/hyprwm/Hyprland/commit/f7fb7e7e49e3b47f9b72c55fbf2d093e1a7981f5)

virtualspan, 2024-08-11:
This issue has been patched in the latest Hyprland stable release.

closed 2025-01-03 by github-actions[bot]