OrcaSlicer/OrcaSlicer #6008: Previously visible 'print flow ratio' invisibly affecting profile but not visible or indicated.

PhilBaz, 2024-07-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

windows 10

### Additional system information

I have just realized that the overall flow ratio that was previously visible in the settings (but has since been removed) has been implicitly affecting my profile. Even though the setting was removed visibly, its value was not reset to one and so the previous setting is essentially just forever frozen there. The only clue that this is even happening is that editing on a per-object basis the print flow ratio is visible in the settings and displays the frozen value that is baked into the profile, but not visible on the global edits.

### Printer

Voron 2.4

### How to reproduce

[image]

### Actual results

Invisible flow ratio affects slice

### Expected results

One would assume that since the setting has been removed that overall flow should be set in the filament profile. However, whatever your value was prior to the removal of the setting from the UI, it will be frozen there also impacting the slicer.

### Project file & Debug log uploads

[Screenshot 2024-07-03 103148.zip](https://github.com/user-attachments/files/16088036/Screenshot.2024-07-03.103148.zip) 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Thanks!

taureau123, 2024-07-26:
Me Too I have the same problem. Any solution ??

PhilBaz, 2024-07-28:
@taureau123 You have to edit the profiles in C:\Users\YOURNAME\AppData\Roaming\OrcaSlicer. Use a text editor and simply remove the whole print flow ratio line. 

This will fix it going forward but the problem will remain in older saved projects.

closed 2024-11-03 by github-actions[bot]