OrcaSlicer/OrcaSlicer #6012: Installation issue OrcaSlicer_Linux_Ubuntu2404_V2.1.1.AppImage

noamix, 2024-07-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer_Linux_Ubuntu2404_V2.1.1.AppImage

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04 fresh install

### Additional system information

# Detalles del informe del sistema
---

## Detalles del informe
- **Fecha generada:**                              2024-07-03 18:34:09

## Información de hardware:
- **Modelo de hardware:**                          ASRock A320M-HDV R4.0
- **Memoria:**                                     16,0 GiB
- **Procesador:**                                  AMD A6-9500 RADEON R5, 8 COMPUTE CORES 2C+6G × 2
- **Gráficos:**                                   NVIDIA GeForce GT 710
- **Capacidad del disco:**                         480,1 GB

## Información de software:
- **Versión de firmware:**                         P3.70
- **Nombre del SO:**                               Ubuntu 24.04 LTS
- **Construcción de SO:**                          (null)
- **Tipo de SO:**                                  64 bits
- **Versión de GNOME:**                            46
- **Sistema de ventanas:**                         X11
- **Versión del núcleo:**                          Linux 6.8.0-36-generic

### Printer

ender 3v2

### How to reproduce

leo@leo-home:~/Descargas$ sudo ./OrcaSlicer.AppImage
[2024-07-03 18:28:33.864734] [0x00007139ebf843c0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:44.924: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:44.963: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:44.965: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:44.968: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:44.970: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:44.973: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:44.977: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:44.983: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:44.987: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
[... 507 lines cut ...]

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:50.089: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:50.089: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:50.098: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:50.099: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:50.105: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:50.109: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6896): Gtk-CRITICAL **: 18:28:50.109: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (orca-slicer:6896): CRITICAL **: 18:28:50.133: Cannot register URI scheme wxfs more than once

** (orca-slicer:6896): CRITICAL **: 18:28:50.133: Cannot register URI scheme memory more than once
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 820x623: Permission denied

### Checklist of files to include

- [ ] Log file
- [ ] Project file

noamix, 2024-07-13:
**follow these steps**

Here is how I fixed it:

Check if you have modesetting enabled via

 cat /sys/module/nvidia_drm/parameters/modeset
If you see printed N then you need to enable it with

 echo options nvidia_drm modeset=1 | sudo tee /etc/modprobe.d/nvidia_drm.conf
Restart your computer

result
./OrcaSlicer.AppImage
[2024-07-12 22:15:07.501260] [0x000074c0200413c0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.295: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.297: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.300: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.302: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.305: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.308: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.311: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.313: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.317: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.319: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.661: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.674: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.674: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.675: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.675: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.678: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.678: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.678: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.678: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.705: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.706: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.932: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.932: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:08.951: gtk_widget_set_size_request: assertion 'width >= -1' failed

[... 182 lines cut ...]
(orca-slicer:6773): Gtk-CRITICAL **: 22:15:12.077: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:12.077: gtk_widget_set_size_request: assertion 'width >= -1' failed
No such file or directory

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:13.493: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:13.493: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:13.510: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:13.513: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:13.526: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:13.531: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:6773): Gtk-CRITICAL **: 22:15:13.532: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (orca-slicer:6773): CRITICAL **: 22:15:13.566: Cannot register URI scheme wxfs more than once

** (orca-slicer:6773): CRITICAL **: 22:15:13.566: Cannot register URI scheme memory more than once
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 820x623: Permission denied
[image]

rakacz, 2024-09-15:
having the same issue :( workaround doesn't work

Robert-Botron, 2024-11-30:
Same issue here, running Fedora 40 with an Nvidia  GTX 770 and the OrcaSlicer_Linux_V2.2.0.AppImage

closed 2025-03-09 by github-actions[bot]