OrcaSlicer/OrcaSlicer #6013: Filament Shrinkage Variant affects Multi Color Printing Walls

HEROofTIME1337, 2024-07-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambu Lab P1s w/ AMS

### How to reproduce

When slicing a multi color model while the selected base filament has a Shrinkage modifier of <100%, colored walls will lose their color. Shrinkage increases the size of the model prior to slicing in order to compensate for filament shrinkage. The below model is has gray filament selected with several faces painted black. 
[image]

### Actual results

The model sliced with the gray filament on a Shrinkage modifier of 99.8% and black filament on 99.6%. As you can see the grey walls have overridden the black walls.
[image]

### Expected results

When sliced with both filaments on 100% Shrinkage (default), black walls are generated normally. 
[image]

### Project file & Debug log uploads

[Barbette and ELX - Barbette.zip](https://github.com/user-attachments/files/16092303/Barbette.and.ELX.-.Barbette.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I suspect it has top do with the object scaling larger and covering the coloring. As if the color mask isn't scaling with the object. Only affects walls because Shrinkage is an XY only scaling, not Z.

fenhelix, 2024-07-07:
Also having this issue on MacOS with @SoftFever 
[image]
[image]
[[a]_MFNano_Remix_Cartridge_Lid_Multibody_Logo.3mf.zip](https://github.com/user-attachments/files/16121112/a._MFNano_Remix_Cartridge_Lid_Multibody_Logo.3mf.zip)

igiannakas, 2024-07-08:
For now disable shrinkage compensation. There is a fix in the works that would attempt to apply scaling to the model as per the latest Prusa release as well but for now, disable shrinkage compensation for multi colour prints and rescale the object manually.

closed 2024-10-15 by github-actions[bot]