OrcaSlicer/OrcaSlicer #6017: Solid infill line in the air affects bridging

vovodroid, 2024-07-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Win10x64

### Printer

Custom

### How to reproduce

Slice attached project (two walls, no infill)

### Actual results

For whatever reason solid infill line is added to the wall:

[image]

And bridge on the next layer fails to connect to the same wall:

[image]

The same happens in Prusa and Bamboo.

### Expected results

Bridge is attached to wall itself.

### Project file & Debug log uploads

[Shape-Box-orca.zip](https://github.com/user-attachments/files/16100941/Shape-Box-orca.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Ataraxia-Mechanica, 2024-07-05:
relevant: https://github.com/SoftFever/OrcaSlicer/issues/5639

vovodroid, 2024-07-05:
Probably, but I don't understand why put this line. Regular bridge to wall connection works good.

Ataraxia-Mechanica, 2024-07-05:
On my machine with this line the bridge can be printed at a higher speed, otherwise it would fail to connect and just fall down, unless printed much slower.

vovodroid, 2024-07-05:
Probably it allows faster bridging on some cases, but in other it causes bridging to fail completely

Ataraxia-Mechanica, 2024-07-05:
Since you are using 0.6 nozzles, can you try disabling thick internal bridges? Might be related since it makes bridges as high as the nozzle width

[image]

vovodroid, 2024-07-05:
Project attached to bug is just example. Real model was printed with thick bridges off

[image]

Ataraxia-Mechanica, 2024-07-05:
Might not be related to my problem of #5639 after all, since my problem is the solid infill itself not sticking and yours is solid infill preventing bridges from sticking.
Also your bridging speed might just be too high for 0.6 nozzles? In the example you posted the speed was 45mm/s and cooling might not be strong enough.

vovodroid, 2024-07-05:
> my problem is the solid infill itself not sticking and yours is solid infill preventing bridges from sticking.

Actually the first causes the later. I just saw failed bridge.

Ataraxia-Mechanica, 2024-07-05:
lol, writing a new issue right now suggesting a new feature that would also fix it not sticking, soon™

closed 2024-10-12 by github-actions[bot]