OrcaSlicer/OrcaSlicer #6020: AppImage crash while accessing printer settings

KWShrimp, 2024-07-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

V2.1.1

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.3 (virginia) - Ubuntu 22.04

### Additional system information

CPU: Ryzen 7 5800x
Memory: 64 gb
GPU: Nvidia 3080 (LHR)

### Printer

Ender 3v2

### How to reproduce

1. Enter Printer Settings
2. Click on any category header (Basic Information, Machine G-Code, Multimaterial, etc.) in advanced mode
3. Switch header

### Actual results

Freeze and crash

### Expected results

Switch category in printer settings

### Project file & Debug log uploads

[debug_Thu_Jul_04_16_52_57_21256.log.0.zip](https://github.com/user-attachments/files/16102844/debug_Thu_Jul_04_16_52_57_21256.log.0.zip)
[Pressure Advance Test.3mf_FILES.zip](https://github.com/user-attachments/files/16102893/Pressure.Advance.Test.3mf_FILES.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Trying to change headers in printer settings gives the JavaScript: Unsupported result type error in the log, however, no window is created that would allow me to click through the error as seen in some previous bug reports. Instead, the appimage just hangs and crashes.

timnolte, 2024-07-07:
I too have had this happen on occasion. It's not consistently all of the time. I'm running the Linux AppImage 2.1.1 on what is essentially Debian 11 in the ChromeOS Linux environment.

henrivdr, 2024-07-12:
This happs to me as well.  But it seems like it is only after switching Printer Profiles.
And mostly when clicking in the gcode tab.

Luke56784, 2024-07-14:
Same issue happens also on Ubuntu 24.04.
[debug_Sun_Jul_14_09_53_10_52784.log.zip](https://github.com/user-attachments/files/16216005/debug_Sun_Jul_14_09_53_10_52784.log.zip)

Same problem with the ubuntu 2404 nightly too.

HakunMatat4, 2024-07-18:
Mint 21.1 Cinnamon

- I can only run OrcaSlicer_Linux_V2.1.1.AppImage nightly release because the WebKit/Glibc update (keep this is mind)
- If I go to Machine g-code to change the offset once, everything works fine
- I could change printer profile at some point before it crashed

So

- if you don't have the need to change printer profile 
- or check printer setting in any way, it works
- Knowing how hell it was for us to have our first Linux release and then webkit/glibc forcing both OrcaSlicer/BambuSlicer to use ubuntu20.04 to generate the AppImages over ubuntu24.04, this problem seems to be related.

vrolife, 2024-08-27:
same problem
* thread #1, name = 'orcaslicer_main', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x18)
  * frame #0: 0x00005555597a4221 orca-slicer`wxWindowBase::RemoveEventHandler(wxEvtHandler*) + 33
    frame #1: 0x000055555691f409 orca-slicer`StateHandler::~StateHandler() + 41
    frame #2: 0x000055555691f42d orca-slicer`StateHandler::~StateHandler() + 77
    frame #3: 0x000055555732bbc4 orca-slicer`TextInput::~TextInput() + 484
    frame #4: 0x00005555597a3409 orca-slicer`wxWindowBase::Destroy() + 57
    frame #5: 0x00005555597a3441 orca-slicer`wxWindowBase::DestroyChildren() + 33
    frame #6: 0x00005555598824a4 orca-slicer`wxWindow::~wxWindow() + 180
    frame #7: 0x0000555556e77d74 orca-slicer`Slic3r::GUI::OG_CustomCtrl::~OG_CustomCtrl() + 164
    frame #8: 0x00005555597a3409 orca-slicer`wxWindowBase::Destroy() + 57
    frame #9: 0x0000555559765cfc orca-slicer`wxSizer::DeleteWindows() + 108
    frame #10: 0x0000555559765cc2 orca-slicer`wxSizer::DeleteWindows() + 50
    frame #11: 0x000055555976639d orca-slicer`wxSizer::Clear(bool) + 189
    frame #12: 0x0000555556d85dbb orca-slicer`Slic3r::GUI::Tab::clear_pages() + 219
    frame #13: 0x0000555556d85e2d orca-slicer`Slic3r::GUI::TabPrinter::clear_pages() + 13
    frame #14: 0x0000555556db7d1b orca-slicer`Slic3r::GUI::Tab::tree_sel_change_delayed(wxCommandEvent&) + 1243
    frame #15: 0x0000555556db804b orca-slicer`wxEventFunctorFunctor<wxEventTypeTag<wxCommandEvent>, Slic3r::GUI::Tab::create_preset_tab()::'lambda9'(wxCommandEvent&)>::operator()(wxEvtHandler*, wxEvent&) + 123
    frame #16: 0x0000555559a2ca41 orca-slicer`wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) + 113
    frame #17: 0x0000555559a2cdbe orca-slicer`wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 126
    frame #18: 0x0000555559a2cf24 orca-slicer`wxEvtHandler::TryHereOnly(wxEvent&) + 36
    frame #19: 0x0000555559a2cc83 orca-slicer`wxEvtHandler::DoTryChain(wxEvent&) + 67
    frame #20: 0x0000555559a2d079 orca-slicer`wxEvtHandler::ProcessEvent(wxEvent&) + 105
    frame #21: 0x000055555733b961 orca-slicer`TabCtrl::sendTabCtrlEvent(bool) + 209
    frame #22: 0x000055555733bb33 orca-slicer`TabCtrl::SelectItem(int) + 339
    frame #23: 0x0000555559a2ca41 orca-slicer`wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) + 113
    frame #24: 0x0000555559a2cdbe orca-slicer`wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 126
    frame #25: 0x0000555559a2cf24 orca-slicer`wxEvtHandler::TryHereOnly(wxEvent&) + 36
    frame #26: 0x0000555559a2cc83 orca-slicer`wxEvtHandler::DoTryChain(wxEvent&) + 67
    frame #27: 0x0000555559a2d079 orca-slicer`wxEvtHandler::ProcessEvent(wxEvent&) + 105
    frame #28: 0x00005555597a32a5 orca-slicer`wxWindowBase::TryAfter(wxEvent&) + 85
    frame #29: 0x000055555731e4d3 orca-slicer`Button::sendButtonEvent() + 195
    frame #30: 0x000055555731e608 orca-slicer`Button::mouseReleased(wxMouseEvent&) + 200
    frame #31: 0x0000555559a2ca41 orca-slicer`wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) + 113
    frame #32: 0x0000555559a2cba3 orca-slicer`wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) + 131
    frame #33: 0x0000555559a2cf4d orca-slicer`wxEvtHandler::TryHereOnly(wxEvent&) + 77
    frame #34: 0x0000555559a2cc83 orca-slicer`wxEvtHandler::DoTryChain(wxEvent&) + 67
    frame #35: 0x0000555559a2d079 orca-slicer`wxEvtHandler::ProcessEvent(wxEvent&) + 105
    frame #36: 0x0000555559a2d9fb orca-slicer`wxEvtHandler::SafelyProcessEvent(wxEvent&) + 11
    frame #37: 0x0000555559886268 orca-slicer`gtk_window_button_release_callback + 376
    frame #38: 0x00007ffff7290bc7 libgtk-3.so.0`___lldb_unnamed_symbol9263 + 103
    frame #39: 0x00007ffff6eb33b0 libgobject-2.0.so.0`g_closure_invoke + 352
    frame #40: 0x00007ffff6ec6076 libgobject-2.0.so.0`___lldb_unnamed_symbol1102 + 2486
    frame #41: 0x00007ffff6ecc42d libgobject-2.0.so.0`g_signal_emit_valist + 1901
    frame #42: 0x00007ffff6eccdbf libgobject-2.0.so.0`g_signal_emit + 143
    frame #43: 0x00007ffff7569844 libgtk-3.so.0`___lldb_unnamed_symbol17645 + 420
    frame #44: 0x00007ffff740783e libgtk-3.so.0`___lldb_unnamed_symbol14201 + 334
    frame #45: 0x00007ffff74092f6 libgtk-3.so.0`gtk_main_do_event + 1862
    frame #46: 0x00007ffff7b13815 libgdk-3.so.0`___lldb_unnamed_symbol2641 + 37
    frame #47: 0x00007ffff7b6d7a2 libgdk-3.so.0`___lldb_unnamed_symbol4040 + 34
    frame #48: 0x00007ffff6db97a9 libglib-2.0.so.0`g_main_context_dispatch + 665
    frame #49: 0x00007ffff6db9a38 libglib-2.0.so.0`___lldb_unnamed_symbol2482 + 520
    frame #50: 0x00007ffff6db9cef libglib-2.0.so.0`g_main_loop_run + 111
    frame #51: 0x00007ffff7408495 libgtk-3.so.0`gtk_main + 117
    frame #52: 0x0000555559868115 orca-slicer`wxGUIEventLoop::DoRun() + 37
    frame #53: 0x000055555997ccd1 orca-slicer`wxEventLoopBase::Run() + 49
    frame #54: 0x000055555994bbb3 orca-slicer`wxAppConsoleBase::OnRun() + 147
    frame #55: 0x00005555599cd4e7 orca-slicer`wxEntry(int&, wchar_t**) + 71
    frame #56: 0x0000555556b2a01b orca-slicer`Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 923
    frame #57: 0x0000555555fc442b orca-slicer`Slic3r::CLI::run(int, char**) + 6219
    frame #58: 0x0000555555ec3209 orca-slicer`main + 329
    frame #59: 0x00007ffff024624a libc.so.6`__libc_start_call_main(main=(orca-slicer`main), argc=1, argv=0x00007fffffffd988) at libc_start_call_main.h:58:16
    frame #60: 0x00007ffff0246305 libc.so.6`__libc_start_main_impl(main=(orca-slicer`main), argc=1, argv=0x00007fffffffd988, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffd978) at libc-start.c:360:3
    frame #61: 0x0000555555fb1eee orca-slicer`_start + 46

closed 2024-12-03 by github-actions[bot]