OrcaSlicer/OrcaSlicer #6022: First purge in prime tower after tool change extremely over sized.

brex480, 2024-07-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1 

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

HP ENVY Laptop 16 12th Gen Intel(R) Core(TM) i7-12700H NVIDIA GeForce RTX 3060 

### Printer

Voron 2.4 Tool Changer

### How to reproduce

Printer is set as a single extruder multimaterial with no cooling tube or ramming. Have tried with flush into prime tower on and off. Tried flush mm3 from 0-100 in flush volume calc, filament minimum purge in filament settings, and prime volume under prime tower settings with the only change is adding to the oversize first filament prime. Have used this setup with previous versions of orca without issue.

### Actual results

Only change when adjusting purge volume in any of the locations is changing the purge/prime that is created at the end of the oversized first prime.

### Expected results

Adjusting the purge/prime mm3 does change the amount extruded as they should except for the very first tool change and prime.

### Project file & Debug log uploads

[Articulated_T-Rex.zip](https://github.com/user-attachments/files/16103366/Articulated_T-Rex.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]
[image]
[image]

53Aries, 2024-07-05:
[image]
Retraction when switching material set to zero for all.

brex480, 2024-07-05:
> ![Screenshot 2024-07-04 205257](https://private-user-images.githubusercontent.com/110708581/345978672-927949cd-3089-4743-b9bc-ded408f2db32.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAxNDY2NDAsIm5iZiI6MTcyMDE0NjM0MCwicGF0aCI6Ii8xMTA3MDg1ODEvMzQ1OTc4NjcyLTkyNzk0OWNkLTMwODktNDc0My1iOWJjLWRlZDQwOGYyZGIzMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwNVQwMjI1NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lZWMyNDlmNTFhNmVmN2U3MWI0NWY5Y2IxZWNmZTg1NTJmNThmZmIzMTBhZWJlZDVhNTc0ODg2ZWM1OWViOGJhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.1UbNBv1eqWiVhPiIF4cYD-izuK1kpXJqE60wn_WRDvU) Retraction when switching material set to zero for all.

They were not set to zero but even setting to zero made no difference. Tested in 1.9.1 as it has been awhile since I have printed multicolor. Imported project with the same settings including having retraction and the prime tower is proper size. 
[image]

igiannakas, 2024-07-08:
Disable precise Z height. 

It doesnt seem to play ball well with the prime tower for some reason. It also shouldn't really be needed for this use case of yours as this is a decorative model and a variance of under +-0.1mm in the Z direction won't make an appreciable difference.

[image]

igiannakas, 2024-07-08:
Also you really dont want arc fitting enabled with a Klipper printer! It makes print quality worse as it converts line paths to arcs and then back to line paths at the klipper end. Not related to this issue but noticed that in your profile.

brex480, 2024-07-08:
> Disable precise Z height.
>
> It doesnt seem to play ball well with the prime tower for some reason. It also shouldn't really be needed for this use case of yours as this is a decorative model and a [...]

Thanks, I just verified by disabling precise Z height on my end and it worked as it should.

closed 2024-10-15 by github-actions[bot]