OrcaSlicer/OrcaSlicer #6024: Strange, unnecessary travel moves ?

Ergonomicmike, 2024-07-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

v2.1.1 portable

### Operating System (OS)

Windows

### OS Version

Win10

### Printer

Klipperized Ender 3 v2

### How to reproduce

Import my Project file
Slice
In the Preview, scrub the toolhead along the first layer

### Actual results

This is not a problem, per se, in the sense that the print prints okay. (Although I could argue that maybe there's some unnecessary oozing caused by the delay during unnecessary travel.) It's more of a puzzle.

But you'll see the tool head move to the front (right) of the zip tie a couple of times. It doesn't appear to flow anything.

### Expected results

No unnecessary travel moves.

### Project file & Debug log uploads

[Strange travel moves.Orca github.ZipTie..zip](https://github.com/user-attachments/files/16104375/Strange.travel.moves.Orca.github.ZipTie.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I've noticed unnecessary travel moves before in some of my other prints. But here it seems the worst. So I post it as an example.

I think this only happens on the first layer. Or maybe that's because I'm usually only watching the first layer?

Ataraxia-Mechanica, 2024-07-05:
You have avoid crossing walls set to 0, so the toolhead wouldn't travel across infill regardless of how long the travel move would be. Set it to something like 40mm could solve the problem.
[image]

closed 2024-07-16 by Ergonomicmike

Trydnt, 2025-01-02:
This is still occuring in 2.2.0, specifically it seems that system cannot easily find the nearest point in the  next loop and so often makes strange travels to other side  of the object before continuing. Terrible stringing.

3d-olympics, 2025-11-28:
I get this as well. In Cura there's an "optimize wall printing order" setting, which helps minimize travel by having the toolhead print nearby walls first.