OrcaSlicer/OrcaSlicer #6027: Slow down for overhanging solid infills

Ataraxia-Mechanica, 2024-07-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

With low infill percentage (<10%), and ensure vertical shell thickness set to "all", on sloped surfaces the solid infill forms large, unsupported overhangs:
[image]

Due to the high speed solid infill is printed at, these overhangs would curl up, generating artifacts on the surface:
(WIP, photo1 here)

One solution for this is to set Don't filter out internal bridges to "no filtering". This works well with higher infill percentages, but due to low infill percentages present here, the bridges generated are long, thus taking up much more time and material. In the simple test case I have, the printing time is increased by +88%, and material usage increased by +52%:
[image]
(WIP, photo2 here)

Another solution is simply use more infill to support the solid infill part. This is particularly effective with gyroid infill, my hypothesis is gyroid infill changes where the solid infill touching the sparse infill vertically, thus preventing the solid infill part from curling up.
[image] 
I used 20% gyroid with infill combination here, at 22.5K acceleration. Increases time by +44%, material usage +41%. But this might not be feasible for certain things that require lightweightness.
(WIP, photo3 here)

Thus I propose adding a detection mechanism that slows down solid infills that form an overhang:
[image]
In my test case, solid infill above 0.6mm is slowed down to 120mm/s from 437mm/s, increasing time only by +11%, and does not increase filament usage.
(WIP, photo4 here)
No artifacts formed on the slope either.

This change should also fix an issue I have, where the soild infill ring before bridging was printed too fast and detaches from the wall, causing bridging failures: 
https://github.com/SoftFever/OrcaSlicer/issues/6017 and https://github.com/SoftFever/OrcaSlicer/issues/5639
(WIP, photo of bridge fail here)

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Slow down for overhanging solid infills.

Ataraxia-Mechanica, 2024-07-05:
Also somehow discovered a bug where the height range modifier wasn't applied on the last plate, no idea how to reproduce it in another file:
[image]

Ataraxia-Mechanica, 2024-07-05:
So, due to general improvements to my printer, 50 degrees can't cause this issue anymore. I'm going to use more extreme angles to make this work

vovodroid, 2024-07-08:
I would also apply bridge cooling to such solid infill.

Ataraxia-Mechanica, 2024-10-14:
no stale pls

closed 2025-01-21 by github-actions[bot]