OrcaSlicer/OrcaSlicer #6028: OrcaSlicer crash on bambulab calibratión dynamic flow if method is "pattern"

Disttrack, 2024-07-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

CPU: Ryzen 5 5600X
RAM: 32GB
GPU: RTX 3070 ti

### Printer

Bambulab P1S

### How to reproduce

1. Go to Calibration
2. Go to Flow Dynamics
3. Go to Manual Calibratión
4. In method, select pattern

[image]

### Actual results

OrcaSlicer crash with "Unhandled unknown exception; terminating the aplication".
[image]

### Expected results

Send calibration to printer

### Project file & Debug log uploads

[logs.zip](https://github.com/user-attachments/files/16107481/logs.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

rmcfadzean, 2024-07-05:
I'm also encountering this issue on Linux. Pattern crashes before anything gets sent to the printer but Line works just fine.

### OrcaSlicer Version
2.1.1

### Operating System (OS)
ArchLinux

### OS Version
Linux soba 6.9.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 28 Jun 2024 04:32:50 +0000 x86_64 GNU/Linux

### Additional system information
CPU: i9-13900H
RAM: 32GB
GPU: RTX 4070 / Raptor Lake-P (Iris Xe Graphics)

### Printer
Bambulab A1 Mini

Mrroot3r, 2024-07-06:
I have been having this issue for a few versions now as well, I want to say it used to work but Im not sure if I always used pattern over line.

OrcaSlicer Version

2.1.1
Operating System (OS)

Windows
OS Version

Windows 10
Additional system information

CPU: Ryzen 5 5900X
RAM: 32GB
GPU: RTX 3090

Printer

Bambulab P1S

Hammerfest, 2024-07-14:
Can confirm, having this issue with the A1 Mini + AMS Lite
Does with all nozzle sizes, havent tested anything but textured bed.
Also tried different options for values, still crashing.

jaeitee, 2024-07-22:
Confirming this issue here as well when choosing Pattern for the calibration.
Mac OSX 14.4, MacBook Pro M2 Max, Orca 2.1.1, Bambu Lab P1S + AMS.

Error log:

> [warning]	2024-07-22 13:03:41.422155[Thread 0x00000001ea527ac0]:01:03:41 pm: Error: OrcaSlicer got an unhandled exception: ConfigOptionFloatOrPercent: Assigning an incompatible type
> [error]	2024-07-22 13:03:41.422175[Thread 0x00000001ea527ac0]:Uncaught exception: ConfigOptionFloatOrPercent: Assigning an incompatible type

PiotrKrzyzek, 2024-09-16:
Same issue here. This has been happening since the last major version, and today I downloaded the new 2.2.0-beta and the issue persists there too.

I likewise think that there are a few related GitHub issues that are more or less the same thing and that can be merged together for easier tracking: 

https://github.com/SoftFever/OrcaSlicer/issues/6249
https://github.com/SoftFever/OrcaSlicer/issues/6128

Maybe others as well.

aarbmx6s, 2024-09-18:
The same issue.
OS: macOS Sonoma 14.14.1 
OrcaSlicer Version: 2.1.1

PR #7248 (2024-10-26): Fix crash when doing PA calib using pattern method from calib tab
Fixes #6028
Fixes #6249
Fixes #7224
closed 2024-10-26 by SoftFever