OrcaSlicer/OrcaSlicer #6035: Would be nice if `Top Interface Layers` was actually `Top interface Thickness` in mm not layer count

tlhintoq, 2024-07-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

This is more of a quality-of-life improvement.  Every time I change layer thickness from .12 to .24 or .36 it means changing the `Top Interface Layers` count to a corresponding value to get the same result.
To be consistent with the other values of this area it would be nice of the interfaces were calculated from thickness.  And so a .5mm interface would auto-calculate from the layer thickness.
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Maybe a dynamic field like line width that accepts both a number (for layer count) or a % of the nozzle (a .4 nozzle could automatically get a 50% interface and thus be a .25 interface thickness), or a mm thickness.  Consistent behavior with other UIX elements.

### Additional context

[image]

Sonerim, 2024-07-06:
What is the platform of your PC? I'm using Orca on Mac OS and seems it's already working as you want.
[image]

tlhintoq, 2024-07-07:
@Sonerim You're looking at shells - actual surfaces of the object.
`Top Interface Layers` is about supports.  Its the interface layer between the tree support and the model.

tlhintoq, 2024-10-06:
Would still love to see this upgrade

tlhintoq, 2025-01-05:
Would be really nice to adjust this.
Changing layer height really shouldn't mean a notepad of related settings to change.
If you could specify `thickness` instead of `layer count` this should happen automatically

hagre, 2025-01-26:
> I agree, this would be better to specify the actual thickness rather than number of layers.

Not always, if you change the layer hight and or nozzle, maybe you like to ensure at least 3 layers on top but the thickness calculats only 2 for example. Best is to give both to a calculation, a minimum number of layers and a minimum thickness in mm . (adaptive layer hights, ... differnet nozzles on multitool ,.. )

hagre, 2025-01-26:
Yes, You are correct. I was thinking on a multimaterial/tool setup (e.g PETG on PLA) where a perfect flat contact surface has to be ensured to garantie good result. (just during writing this, i was thinking on a feature request for ironing the contact surface of the support)

tlhintoq, 2025-07-11:
Nudge