OrcaSlicer/OrcaSlicer #6040: Color Swaps should affect Print by Object

Kingsidorak, 2024-07-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Print by Object not being as smart as it should be

If I have 3 Green parts and 1 Lime: When Print by Object Auto Arrange is active, the order is Green, Lime, Green, Green

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

clicking arrange when Print by Object is active, and multiple colors are on the selected place, then it should attempt to arrange by least color swaps

### Describe alternatives you've considered

arranging models myself which doesn't seem to change the order in which parts are printed........???

Noisyfox, 2024-07-09:
> arranging models myself which doesn't seem to change the order in which parts are printed........???

That's correct. Reorder the object list manually instead.

Kingsidorak, 2024-07-09:
Just moving them around the buildplate doesn't seem to change the print sequence which is the main issue for me. It seems like I have to move a part off the plate, and then add it on, or be very mindful of the order in which I place them in, which seems like unintended function

Noisyfox, 2024-07-09:
Use the object list.

closed 2024-10-15 by github-actions[bot]