OrcaSlicer/OrcaSlicer #6043: Appimage device tab unable to display Mainsail or Fluidd

masterjuggler, 2024-07-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Fedora 40

### Printer

Custom corexy running klipper

### How to reproduce

1. Install webkit2gtk
2. Add WEBKIT_DISABLE_DMABUF_RENDERER=1 env variable
3. Run Appimage
4. Add printer local ip to network settings and test connection
5. Go to device tab

### Actual results

Mainsail displays Initializing bar and freezes. Fluidd Makes it to an interface and freezes before all elements are loaded.

### Expected results

Klipper interface (Mainsail or Fluidd) displays.

### Project file & Debug log uploads

[debug_Sat_Jul_06_09_50_10_991747.txt](https://github.com/user-attachments/files/16116825/debug_Sat_Jul_06_09_50_10_991747.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

I can access both webui's no problem in a browser, and I can access another web page i.e. google.com in orcaslicer. I am able to upload gcode files from orcaslicer. The video in the device tab also runs when I have no local ip entered in the printer's network settings.

masterjuggler, 2024-07-07:
> See [elementary/browser#104 (comment)](https://github.com/elementary/browser/discussions/104#discussioncomment-8654515) for a possible workaround.

I actually already have nvidia-drm.modeset=1 passed as a kernel parameter from when I set up gpu passthrough VFIO, and I double checked it is set in /sys/module/nvidia_drm/parameters/modeset.

If it is somewhere upstream, how do I even go about finding the source of this issue?

masterjuggler, 2024-07-07:
Just so we're on the same page about what exactly is happening, the device tab does seem to load and initially connect, but then freezes when it tries to load the UI. I added a couple of clips showing the exact behavior:

[image]
[image]

masterjuggler, 2024-07-07:
Just grabbed the latest nightly and see no difference in behavior unfortunately. Going back to older versions like 1.9.0 and 1.7.0 doesn't seem to help either.

masterjuggler, 2024-07-08:
Other than needing to install webkit2gtk4.1 to run it, unfortunately no change in behavior.

hgelpke, 2024-07-09:
Experiencing the same issue on Pop OS using a 3060. Device tab works when I power on but when I do a firmware restart it can't be accessed.

Using the ip of my printer with no authentication via l/p or API

masterjuggler, 2024-07-10:
\*Sigh\* I also just spun up a fresh fedora 40 VM and couldn't reproduce the issue. Sounds like it is something to do with my setup, nvidia or otherwise. Is there a decent way to go about troubleshooting this? The logs I've seen so far for orcaslicer haven't included information relevant to this.

hgelpke, 2024-07-10:
> *Sigh* I also just spun up a fresh fedora 40 VM and couldn't reproduce the issue. Sounds like it is something to do with my setup, nvidia or otherwise. Is there a decent way to go about troubleshooting this? The logs I've seen so far for orcaslicer haven't included information relevant to this.

The common thread so far seems to be Nvidia since I'm assuming the spun up VMs are just using the generic linux display drivers and nothing installed after that.

tomoinn, 2024-07-30:
I'm seeing the same with the nightly builds and compiling myself using webkit2gtk4.1 to match Ubuntu 24.04 (linux mint 22), with a 4080 using the nvidia drivers. I have to set WEBKIT_DISABLE_DMABUF_RENDERER=1 to get the home and project pages to load at all, and sometimes the device tab loads but immediately freezes, sometimes it doesn't load at all.

This also applies building with the earlier webkit2gtk4.0, exactly the same behaviour. I'm assuming some problem between webkit and nvidia drivers, but it's kind of hard to know from what I'm seeing.

tomoinn, 2024-07-31:
Weirdly, the *only* site that doesn't render properly is the mainsail interface. Setting the 'printer UI' address to the BBC news (and playing video) works fine, as do the webGL example galleries, so it's a bug triggered particularly by how mainsail works. Of course, that's the only site we actually need to render here

tomoinn, 2024-07-31:
Setting the printer UI to one of the sub-pages within mainsail produces variable results. The 'machine' tab loads and is showing live temperatures and allowing scrolling (scrollwheel only, clicking on the scroll bar does not work) but no interactivity . Keyboard input does appear to work, I can select the different page tabs but not consistently.

keteflips, 2024-07-31:
Here we go I cant use this feature since... I can remember, that's a disaster...

karaktaka, 2024-10-30:
Looks like this is still an issue. Latest build v2.2.0 still needs these env variables to be usable

closed 2025-02-06 by github-actions[bot]