OrcaSlicer/OrcaSlicer #6045: Orcaslicer Shutting down when you try to print a gcode from micro sd

Hellclay, 2024-07-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambulab X1C/P1P

### How to reproduce

device->micro-sd->Model->print

### Actual results

Orcaslicer shuts down

### Expected results

Print Window should open to select Timelapse,Flow Calibration, Bed Leveling and AMS Slot. In bambustudio that Problem doesnt exist.

### Project file & Debug log uploads

[crash_Fri_Jul_05_18_54_34_0.log](https://github.com/user-attachments/files/16117960/crash_Fri_Jul_05_18_54_34_0.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Haleysoftware, 2024-07-07:
I also just encountered this bug today. I had no issues with starting a print from SD card yesterday or otherwise in the past.

Haleysoftware, 2024-07-07:
I just tried to print that file from the printer's (P1S) display and noticed it was a 3mf file and not a gcode file. I used the "Send" option in Orca Slicer to send that file to the printer SD card yesterday. I just deleted the file and sent it again after slicing it. The sent file is still a 3mf.

I deleted the file from the SD card again and tried the "Print Plate" option. The printer started the print but when I checked the cached file, it was a 3mf. It started printing without issue and even did the start gcode. I don't know what is up with this file.

Here is the file that is causing the issue for me.

[s2tray+AC.3mf.zip](https://github.com/user-attachments/files/16119773/s2tray%2BAC.3mf.zip)

PewnyPL, 2024-07-07:
Also encountered that issue today, I checked and none ofthe files I have on the SD card can be chosen to print without crashing. Here are the log files (Windows machine).
[crash_Sun_Jul_07_17_35_13_0.log](https://github.com/user-attachments/files/16119852/crash_Sun_Jul_07_17_35_13_0.log)
[debug_Sun_Jul_07_17_34_49_65884.log](https://github.com/user-attachments/files/16119855/debug_Sun_Jul_07_17_34_49_65884.log)

closed 2024-10-14 by github-actions[bot]