OrcaSlicer/OrcaSlicer #6054: Raft: Doesn't always print perimeter first when using tree supports

ReinhardC, 2024-07-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

CPU: AMD Threadripper, probably not important for this issue

### Printer

BambLab A1 Mini

### How to reproduce

1) New Project (Bambu Lab A1 Mini, Textured PEI Plate, 0.2mm Standard @BBL A1M)
2) Add Benchy Model
3) Activate Supports (Tree `Hybrid`, `Strong `or `Slim` - - - Tree `Organic` causes no issue!)
4) Set Raft Layers to 2
5) Slice, observe first layer printing order

### Actual results

For some of the raft islands, Orca prints the inside of the islands first, then the perimeter. This can cause print failures because the raft sometimes comes off the build plate when quickly changing directions when it can't attach to the perimeter.

### Expected results

Print the outline of each island/all the islands first 

### Project file & Debug log uploads

Very easy to reproduce, so probably no log/project required?

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

[image]

closed 2024-10-14 by github-actions[bot]

Saylux, 2026-09-18:
PLEASE FIX THIS! This has caused me constant pains and headaches for years! this is the biggest reason I hate orca slicer versus cura. I wish all the new good printers used Cura instead of this half baked software.

its-astro, 2026-10-20:
I have also experienced this issue on a P1S, toggling off tree supports or using organic tree support fixes this and causes the normal expected behavior.