OrcaSlicer/OrcaSlicer #6056: Not all z-hops is diagonal when selecting option "slope" in z-hops.

morgan55555, 2024-07-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Snapmaker A350

### How to reproduce

1. Generate any g-code file with sloped z-hops. It's better to set big z-hop value, just for easily finding them, but even with default retracts problem is persist.
2. Enable showing printer moves and inspect it and it's gcode.
3. Find z-hops, which moving up without slope, causing big stringins issue on PETG.

### Actual results

This z-hope is moving up first, making a big issue for my PETG.

[image]

### Expected results

All z-hopes should be sloped, not only a half.

### Project file & Debug log uploads

[Retraction test bugged.zip](https://github.com/user-attachments/files/16122343/Retraction.test.bugged.zip)
[log.zip](https://github.com/user-attachments/files/16122344/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Rickthebig, 2024-07-17:
@morgan55555 
Since I am engaged in tuning optimal settings against PETG stringing I also tested slope z hop and noticed that the angle at which this type of z hop is executed is quite steep, leading to stringing. This defeats its purpose. The slope angle should be flat. Actually I was assuming that the slope z hop moves the nozzle directly above the next extrusion spot, this way the angle would still be flat enough to wipe the nozzle and at the same time perform a z lift comprising the whole next travel.

Until now, I am not able to get rid of stringing by means of using z hop of any kind in Orca. Without z hop, I get no stringing but then I get blobs along the travel path when crossing the border to the next extrusion spot, which would need a... z hop to be avoided - catch22. I actually need to z hop at the beginning of the travel move, but a z hope at its end. The solution is a diagonal straight travel move.

Just in case I am hit by the same bug, would you mind to share your settings that are supposed to avoid stringing with PETG?
Specifically:
1. Retraction length and speed
2. Z hop height
3. Layer Width
4. Wipe on loops set?
5. Wipe before external loops set?
6. Role based wipe speed set?
7.   (if no:) % of wipe speed
8. Wipe distance
9. Retract amount before wipe (%)

This bug will possibly not be fixed soon. In the meantime you can use this online service to postprocess your gcode files by inserting a "diagonal z hop":
https://teachingtechyt.github.io/diagonalZhop.html
I tested it, it works.

Thanks in advance.

morgan55555, 2024-07-18:
My settings is all default, except retraction length/speed (filament specific, PETG is needed more retraction length and speed), and enabled slopes in printer, not in filament, then stringing is not eliminated completely, but there is much less amount of strings that before, and you can use hot air for finishing.
In my case only diagonal slope worked well, spiral is stringing too.
Also, I uploaded my project file, you can load it and view all of my settings.

Rickthebig, 2024-07-18:
> In my case only diagonal slope worked well, spiral is stringing too.

There is a proposition of a "half slope z hop":
https://github.com/SoftFever/OrcaSlicer/discussions/3439
Would be the best solution I think. But developers don't care I think.

closed 2024-10-25 by github-actions[bot]